riscv-software-src/riscv-tools #237: The problem when running a progam using spike.

yonghaoXuGit, 2018-08-30:
I compile a helloworld program using riscv64-unknown-linux-gnu-gcc,and get the executable file,then run "spike hello".But I got an error:"ERROR: invalid load from debug module: 8 bytes at 0x0000000000010538
terminate called after throwing an instance of 'trap_load_access_fault'".How to solve it?Here is my source code:
#include <stdio.h>
int main() {
        printf("abc");
        return 0;
}

yonghaoXuGit, 2018-08-30:
So,what's the difference between riscv64-unknown-elf-gnu-gcc and riscv64-unknown-linux-gnu-gcc,when i check the executable files they generated,the size of these two files is very different.Why?

And you say that spike is designed to run linux kernel,why can't it run linux programs?When I disassemble the executable file compiled by riscv64-unknown-linux-gnu-gcc,it's instruction is risc-v style.So why can't run with spike?I have been confused about this for a long time.But don't know why.

jim-wilson, 2018-08-31:
When a linux program wants to do I/O, it makes a system call, which generates a trap, which assumes that there is an OS available to service the trap.  When the linux kernel wants to do I/O, it talks directly to hardware via device drivers.  These are two very different things.  For the first one to work, the simulator must emulate an OS.  Spike does not do that, but riscv-pk can to some extent.  For the second one to work, the simulator must emulate hardware devices.  Spike does have a few hardware devices that it emulates.

The main difference between the elf and linux compilers is that they use different C libraries.  The linux compiler defaults to glibc, which is assumes it is running on top of a linux kernel.  The elf compiler uses newlib and its companion library libgloss, which assume that they are running on bare hardware.  The two libraries support the same ABI and basic API, but glibc has many more features than newlib.  The linux compiler produces code that will only run on a linux system.  The elf compiler produces code that can run on various systems, though some things like I/O may or may not be supported depending on where and how you run it.

Some of this stuff is basic info about how operating systems work.  You might try reading a book about operating system design and implementation to learn more.

yonghaoXuGit, 2018-08-31:
OK,thanks,maybe I should know more about Operating System.