riscv-software-src/riscv-tools #243: Boot Linux failed:Cannot open root device "generic-blkdev" or unknown-block(0,0): error -6

yonghaoXuGit, 2018-09-13:
When I user command "spike bbl vmlinux" to boot linux,I encountered the error:"Cannot open root device "generic-blkdev" or unknown-block(0,0): error -6",and stuck at：“ret_from_syscall+0xc/0x10”. What's the reason of this problem, I followed the steps in tutorial in github.

aswaterman, 2018-09-13:
Unfortunately, the instructions in the README are vastly out of date (note they were last updated over four years ago).  I've submitted a pull request to redact the outdated README.  In the mean time, I advise looking at SiFive's Freedom-U SDK (https://github.com/sifive/freedom-u-sdk) or soliciting help on the RISC-V software development mailing list (https://riscv.org/mailing-lists/).

yonghaoXuGit, 2018-09-13:
OK,thanks!

closed 2018-09-13 by aswaterman