riscv-software-src/riscv-tools #247: hello.c:1:10: fatal error: stdio.h: No such file or directory

hepaad, 2018-09-22:
Hi 
I finished to install the riscv-tool and run everything fine. but
when I try to run the command.
**riscv64-unknown-elf-gcc -o hello hello.c**
I have the follow error.

root@hepaad-Vostro-5470:~/Documentos/old/ARCH# riscv64-unknown-elf-gcc -o hello hello.c
hello.c:1:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.

I hope you can help me. and thanks.

hepaad, 2018-09-22:
/home/hepaad/Documentos/ARCH/riscv-tools/riscv-tests/build/../benchmarks/dhrystone/dhrystone_main.c:326:1: warning: return type defaults to 'int' [-Wimplicit-int]
 Proc_5 () /* without parameters */
 ^~~~~~
Installing project riscv-tests

RISC-V Toolchain installation completed!
root@hepaad-Vostro-5470:~/Documentos/ARCH/riscv-tools# echo $RISCV
/home/hepaad/Documentos/ARCH/riscv
root@hepaad-Vostro-5470:~/Documentos/ARCH/riscv-tools# riscv64-unknown-elf-gcc -o hello hello.c
riscv64-unknown-elf-gcc: error: hello.c: No such file or directory
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.
root@hepaad-Vostro-5470:~/Documentos/ARCH/riscv-tools# cd ..
root@hepaad-Vostro-5470:~/Documentos/ARCH# riscv64-unknown-elf-gcc -o hello hello.c
hello.c:1:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.
root@hepaad-Vostro-5470:~/Documentos/ARCH# 
 Details when finished the installation..

jim-wilson, 2018-09-23:
riscv-tools has an old obsolete unmtainined and known broken copy of the toolchain.  You shouldn't use it.  You should use riscv-gnu-toolchain directly instead.  But that isn't directly related to your problem.

Something went wrong with your toolchain build and install.  You will have to try to debug the build problem.  Do you have header files in your toolchain install tree?  Is gcc searching the right directories?  You can see the directories gcc is searching by adding a -v option to the compile command.  In your install tree, you should see a riscv64-unknown-elf/include/stdio.h file.

hepaad, 2018-09-23:
Thank for you answer, Ok I will change the reference in future only riscv-gnu-toolchain. (Do you have a reference manual of installation for this?.

Well, I send you the information that my computer show wiht -v.  

hepaad@hepaad-Vostro-5470:~/Documentos/ARCH$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/riscv64-unknown-elf/8.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/hepaad/Documentos/ARCH3/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/usr/local --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-system-zlib --with-newlib --with-sysroot=/usr/local/riscv64-unknown-elf --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=.././riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 8.2.0 (GCC) 

I saw that folder ARCH3, I deleted myabe it is problem.  

How I can re-install this program?

Regards
Adolfo Hernandez.

jim-wilson, 2018-09-24:
riscv-gnu-toolchain has a README.md file.

You need to use -v when compiling, e.g. riscv64-unknown-elf-gcc -v hello.c, and then look at the header file include paths in the output.

The directory where you built gcc is not required to run gcc after installing, as long as you did not install into the build tree.  This does make it harder to debug build problems if you don't have the build tree anymore though.

hepaad, 2018-09-24:
Hi Jim: 
Check my log file. 
hepaad@hepaad-Vostro-5470:~/programas$ riscv64-unknown-elf-gcc -v hello.c
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/riscv64-unknown-elf/8.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/hepaad/Documentos/ARCH3/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/usr/local --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-system-zlib --with-newlib --with-sysroot=/usr/local/riscv64-unknown-elf --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=.././riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 8.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-march=rv64imafdc' '-mabi=lp64d'
 /usr/local/libexec/gcc/riscv64-unknown-elf/8.2.0/cc1 -quiet -v hello.c -quiet -dumpbase hello.c -march=rv64imafdc -mabi=lp64d -auxbase hello -version -o /tmp/ccWJd5bg.s
GNU C17 (GCC) version 8.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/usr/local/riscv64-unknown-elf/usr/local/include"
ignoring nonexistent directory "/usr/local/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/include"
ignoring nonexistent directory "/usr/local/riscv64-unknown-elf/usr/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/lib/gcc/riscv64-unknown-elf/8.2.0/include
 /usr/local/lib/gcc/riscv64-unknown-elf/8.2.0/include-fixed
End of search list.
GNU C17 (GCC) version 8.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: a876a06a95b9c0558b0c192b502929a7
hello.c:1:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.

Regards
Adolfo Hernandez.

hepaad, 2018-09-24:
H jim:

I copied the folder that misising in the official route and working fine. 

Thank you for you help. 

Only one question do you know a recipe to create a new isntruction like mac,  
a=+ b*c;

jim-wilson, 2018-09-24:
On Sun, Sep 23, 2018 at 7:47 PM hepaad <notifications@github.com> wrote:

> Only one question do you know a recipe to create a new isntruction like
> mac,
>
> a=+ b*c;
>

See the second message in the thread at
https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/sL_OHXYj3LY

Jim

Ray-Lei-93, 2019-12-15:
> H jim:
> 
> I copied the folder that misising in the official route and working fine.
> 
> Thank you for your help.
> 
> Only one question. do you know a recipe to create a new isntruction? like mac,
> a=+ b*c;

Hi hepaad,

Can you tell me where I can copied those files form? I don't know where to find it.

hepaad, 2019-12-17:
Hi Lei:

I don't remember where is it. sorry.

Regards
Hepaad.

jim-wilson, 2019-12-17:
In the source tree, for an embedded elf build, the header files should be in riscv-newlib/newlib/libc/include.  But these files should automatically be installed as part of the build.

Ray-Lei-93, 2019-12-17:
> […](#)
> -- Adolfo Hernández Padilla Ing. de Pruebas

Thanks for your reply! I find it :)

Ray-Lei-93, 2019-12-17:
Hi Jim,

Thanks a lot! I find the header files are in the right directory.
The only problem is that I try to run the following command,
`riscv64-unknown-linux-gnu-gcc -o hello hello.c`.
Seems like I misunderstand the meaning of *Linux cross-compiler* as the cross-compiler run under linux.

jim-wilson, 2019-12-17:
I pointed at the newlib header files used by the embedded elf compiler.  If you are using a linux target compiler then you need the glibc header files instead of the newlib header files.  But yes, there are two possible cross compilers here, one that produces code for embedded elf targets, and one that produces code for linux targets.

ali-tah, 2022-07-13:
I have the same problem on Windows.
(gcc -c main.cpp -o main.o
main.cpp:1:19: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
                   ^
compilation terminated.
Makefile:17: recipe for target 'main.o' failed
mingw32-make: *** [main.o] Error 1 )