riscv-software-src/riscv-tools #248: python PyImport_FixupBuiltin error when compiling on linux

michaelbarrett, 2018-09-23:
I am building this inside rocket chip, here are the steps:

git submodule update --init --recursive
$ export RISCV=/home/heyoka/Prog/rocket-chip/riscv-tools
$ export PATH="${PATH}:$RISCV/bin"
$ ./build.sh

this is arch linux, i believe i have all the necessary packages installed.
Last couple of lines of the error:

```
In file included from /home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/defs.h:785,
                 from /home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/python/py-xmethods.c:20:
/home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/utils.h:74:24: warning: unnecessary parentheses in declaration of ‘make_cleanup_free_section_addr_info’ [-Wparentheses]
 extern struct cleanup *(make_cleanup_free_section_addr_info
                        ^
In file included from /home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/defs.h:785,
                 from /home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/python/python.c:20:
/home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/utils.h:74:24: warning: unnecessary parentheses in declaration of ‘make_cleanup_free_section_addr_info’ [-Wparentheses]
 extern struct cleanup *(make_cleanup_free_section_addr_info
                        ^
/home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/python/python.c: In function ‘bool do_start_initialization()’:
/home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/python/python.c:1584:45: error: too few arguments to function ‘int _PyImport_FixupBuiltin(PyObject*, const char*, PyObject*)’
   _PyImport_FixupBuiltin (gdb_module, "_gdb");
                                             ^
In file included from /usr/include/python3.7m/Python.h:126,
                 from /home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/python/python-internal.h:94,
                 from /home/heyoka/Prog/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/python/python.c:94:
/usr/include/python3.7m/import.h:115:17: note: declared here
 PyAPI_FUNC(int) _PyImport_FixupBuiltin(
                 ^~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:1947: python.o] Error 1
make[2]: *** [Makefile:10420: all-gdb] Error 2
make[1]: *** [Makefile:849: all] Error 2
make: *** [Makefile:335: stamps/build-binutils-newlib] Error 2
```

jim-wilson, 2018-09-23:
Another thing I should mention, gdb is currently a mess because it is half upstream.  We have a port in the FSF gdb tree, which is a different port than the one in the riscv-gnu-toolchain tree, and some things work better in the FSF tree and some things work better in the riscv-gnu-toolchain tree.  Until this gets sorted out, which could be many more months, riscv-gnu-toolchain will continue to have an old version of gdb that doesn't have the python-3.7 support.  The arch distro tends to have the very newest versions of things, which does not work well with the current state of the RISC-V tools ecosystem, which is tied to older systems.  SiFive uses Ubuntu 16.04 for instance.  You will have better luck if you use a different distro, or if you start using upstream sources for the GNU toolchain, instead of riscv-tools and riscv-gnu-toolchain.

ferdous313, 2018-10-03:
I have got the same error. I am using gdb 8.2 and python 3.7. I found in the suggestion that it works well with gbd 8.2 but I got those errors.

mada0:~/boom-template/rocket-chip/emulator$gdb 
GNU gdb (GDB) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.

ferdous313, 2018-10-08:
Thank you very much for your suggestions.
I do gitclone from this link the risc-binutils-gdb folder (https://github.com/bminor/binutils-gdb).
I got rid of the python issue. 
This one is giving this new error.. Please help me solve this.
./scripts/build-tools.sh 
output:
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/complete.c: In function 'fnwidth':
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/complete.c:701:12: warning: implicit declaration of function 'wcwidth'; did you mean 'fnwidth'? [-Wimplicit-function-declaration]
        w = wcwidth (wc);
            ^~~~~~~
            fnwidth
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/display.c: In function 'rl_redisplay':
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/display.c:768:15: warning: implicit declaration of function 'wcwidth' [-Wimplicit-function-declaration]
        temp = wcwidth (wc);
               ^~~~~~~
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/mbutil.c: In function '_rl_find_next_mbchar_internal':
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/mbutil.c:122:12: warning: implicit declaration of function 'wcwidth' [-Wimplicit-function-declaration]
        if (wcwidth (wc) == 0)
            ^~~~~~~
configure: WARNING: libipt is missing or unusable; some features may be unavailable.
configure: WARNING: babeltrace is missing or unusable; GDB is unable to read CTF data.
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/c-exp.y: warning: 42 shift/reduce conflicts [-Wconflicts-sr]
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/c-exp.y: warning: 53 reduce/reduce conflicts [-Wconflicts-rr]
 /soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/m2-exp.y: warning: 34 shift/reduce conflicts [-Wconflicts-sr]
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/m2-exp.y:301.25-44: warning: rule useless in parser due to conflicts [-Wother]
                         { start_arglist(); }
                         ^^^^^^^^^^^^^^^^^^^^
Unescaped left brace in regex is deprecated here (and will be fatal in Perl 5.32), passed through in regex; marked by <-- HERE in m/^\s+@([[:alnum:]][[:alnum:]\-]*)({ <-- HERE })?\s*/ at /usr/share/texinfo/Texinfo/Parser.pm line 5481.
Unescaped left brace in regex is deprecated here (and will be fatal in Perl 5.32), passed through in regex; marked by <-- HERE in m/^\s+@([[:alnum:]][[:alnum:]\-]*)({ <-- HERE })?\s*(\@(c|comment)((\@|\s+).*)?)?/ at /usr/share/texinfo/Texinfo/Parser.pm line 5485.
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/doc/python.texi:2907: @node seen before @end defun
/soe/nferdous/boom-template/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/doc/python.texi:2868: @pxref reference to nonexistent node `Frames in Python'
make[6]: *** [Makefile:486: gdb.info] Error 1
make[5]: *** [Makefile:2004: subdir_do] Error 1
make[4]: *** [Makefile:1738: install-only] Error 2
make[3]: *** [Makefile:1732: install] Error 2
make[2]: *** [Makefile:10449: install-gdb] Error 2
make[1]: *** [Makefile:2225: install] Error 2
make: *** [Makefile:336: stamps/build-binutils-newlib] Error 2
A error has been encountered while building the toolchain.

ferdous313, 2018-10-08:
This helped solve the above issue
git clone git://sourceware.org/git/binutils-gdb.git 
Please help me solve these errors:
../../../riscv-gcc/gcc/gimplify.c: In function 'void gimplify_adjust_omp_clauses(gimple**, gimple_seq, tree_node**, tree_code)':
../../../riscv-gcc/gcc/gimplify.c:8836:37: warning: bitwise comparison always evaluates to true [-Wtautological-compare]
        && (n->value & GOVD_WRITTEN) != 1)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In function 'void ivs_params_clear(ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:98:7: error: 'isl_id_free' was not declared in this scope
       isl_id_free (it->first);
       ^~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:98:7: note: suggested alternative: 'isl_aff_free'
       isl_id_free (it->first);
       ^~~~~~~~~~~
       isl_aff_free
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'tree_node* translate_isl_ast_to_gimple::gcc_expression_from_isl_ast_expr_id(tree, isl_ast_expr*, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:293:3: error: 'isl_id_free' was not declared in this scope
   isl_id_free (tmp_isl_id);
   ^~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:293:3: note: suggested alternative: 'isl_aff_free'
   isl_id_free (tmp_isl_id);
   ^~~~~~~~~~~
   isl_aff_free
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'loop* translate_isl_ast_to_gimple::graphite_create_new_loop(edge, isl_ast_node*, loop_p, tree, tree, tree, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:629:5: error: 'isl_id_free' was not declared in this scope
     isl_id_free (res->first);
     ^~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:629:5: note: suggested alternative: 'isl_aff_free'
     isl_id_free (res->first);
     ^~~~~~~~~~~
     isl_aff_free
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'edge_def* translate_isl_ast_to_gimple::translate_isl_ast_for_loop(loop_p, isl_ast_node*, edge, tree, tree, tree, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:669:53: error: 'isl_id_get_user' was not declared in this scope
       ast_build_info *for_info = (ast_build_info *) isl_id_get_user (id);
                                                     ^~~~~~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:669:53: note: suggested alternative: 'isl_arg_user'
       ast_build_info *for_info = (ast_build_info *) isl_id_get_user (id);
                                                     ^~~~~~~~~~~~~~~
                                                     isl_arg_user
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:672:7: error: 'isl_id_free' was not declared in this scope
       isl_id_free (id);
       ^~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:672:7: note: suggested alternative: 'isl_aff_free'
       isl_id_free (id);
       ^~~~~~~~~~~
       isl_aff_free
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'edge_def* translate_isl_ast_to_gimple::translate_isl_ast_node_user(isl_ast_node*, edge, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:876:31: error: 'isl_id_get_user' was not declared in this scope
   poly_bb_p pbb = (poly_bb_p) isl_id_get_user (name_id);
                               ^~~~~~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:876:31: note: suggested alternative: 'isl_arg_user'
   poly_bb_p pbb = (poly_bb_p) isl_id_get_user (name_id);
                               ^~~~~~~~~~~~~~~
                               isl_arg_user
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:882:3: error: 'isl_id_free' was not declared in this scope
   isl_id_free (name_id);
   ^~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:882:3: note: suggested alternative: 'isl_aff_free'
   isl_id_free (name_id);
   ^~~~~~~~~~~
   isl_aff_free
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In function 'isl_id* ast_build_before_for(isl_ast_build*, void*)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2886:19: error: 'isl_space_dim' was not declared in this scope
   int dimension = isl_space_dim (schedule_space, isl_dim_out);
                   ^~~~~~~~~~~~~
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2886:19: note: suggested alternative: 'isl_aff_dim'
   int dimension = isl_space_dim (schedule_space, isl_dim_out);
                   ^~~~~~~~~~~~~
                   isl_aff_dim
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2890:3: error: 'isl_space_free' was not declared in this scope
[... 10 lines cut ...]
   isl_id *id = isl_id_alloc (isl_ast_build_get_ctx (build), "", for_info);
                ^~~~~~~~~~~~
                isl_mat_alloc
In file included from ../../../riscv-gcc/gcc/hash-table.h:236,
                 from ../../../riscv-gcc/gcc/coretypes.h:369,
                 from ../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:29:
../../../riscv-gcc/gcc/vec.h: In instantiation of 'bool vec<T, va_heap, vl_ptr>::reserve(unsigned int, bool) [with T = std::pair<gphi*, gphi*>]':
../../../riscv-gcc/gcc/vec.h:1554:3:   required from 'T* vec<T, va_heap, vl_ptr>::safe_push(const T&) [with T = std::pair<gphi*, gphi*>]'
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:1860:72:   required from here
[... cut ...]
                 from ../../../riscv-gcc/gcc/system.h:227,
                 from ../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:28:
/usr/include/c++/8.2.1/bits/stl_pair.h:198:12: note: 'struct std::pair<gphi*, gphi*>' declared here
     struct pair
            ^~~~
make[2]: *** [Makefile:1099: graphite-isl-ast-to-gimple.o] Error 1
make[1]: *** [Makefile:4220: all-gcc] Error 2
make: *** [Makefile:352: stamps/build-gcc-newlib-stage1] Error 2
A error has been encountered while building the toolchain.
mada0:~/boom-template$

ferdous313, 2018-10-08:
Error:
./../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2891:16: error: 'isl_id_alloc' was not declared in this scope
isl_id id = isl_id_alloc (isl_ast_build_get_ctx (build), "", for_info);
isl_aff_dim
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2890:3: error: 'isl_space_free' was not declared in this scope
isl_space_free (schedule_space);

jim-wilson, 2018-10-08:
It looks like you are trying to build old software on a new OS.  That isn't going to work, as you will end up hitting a lot of versioning problems.  The easy solution is to use an older OS that is compatible with the software that you are trying to build.  Or alternatively, you could stop using anything from github.com/riscv and just use the FSF tools directly.  The FSF versions of the tools should work on a new OS.  However, it looks like you are trying to build the hardware packages for boom, and that may not work with current FSF tools, in which case using an older OS may be the only workable solution.  I know nothing about boom, but SiFive internally is using Ubuntu 16.04 and the default versions of the various riscv tools are known to work on this OS.

The gcc error is because you have an old version of gcc and a new version of the isl library that aren't compatible with each other.  Either you need to upgrade gcc or downgrade isl.  Or alternatively, you might be able to build gcc by configuring without the isl support.  You could try adding --without-isl to the gcc configure options.