riscv-software-src/riscv-tools #249: make output/rv64ui-p-simple.out:Error

ferdous313, 2018-09-29:
Hi,
I am getting error when I am trying to run a test as
make output/rv64ui-p-simple.out
Error message:
make[1]: Leaving directory '/mada/users/nferdous/boom-template/verisim/generated-src/boom.system.TestHarness.BoomConfig'
make: *** No rule to make target '/soe/nferdous/rocket-chip/riscv-tools/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple', needed by 'output/rv64ui-p-simple'.  Stop
Any suggestion will be appreciated.

ferdous313, 2018-09-29:
Hi,
I am getting error when I am trying to run a test in /boom-template/verisim
/boom-template/verisim$pwd
boom-template/verisim
/boom-template/verisim$ make output/rv64ui-p-simple.out
Error message:
make[1]: Leaving directory '/mada/users/nferdous/boom-template/verisim/generated-src/boom.system.TestHarness.BoomConfig'
make: *** No rule to make target '/soe/nferdous/rocket-chip/riscv-tools/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple', needed by 'output/rv64ui-p-simple'. Stop
Any suggestion will be appreciated.

palmer-dabbelt, 2018-10-01:
This doesn't look like a riscv-tools bug, but instead is probably some sort of mismatch between your test harneess and riscv-tools.  @ccelio would be the best bet here, as it appears you're using BOOM.  You should try to make sure all your versions match correctly, as that's usually the issue.

I'm going to close this, as it's not a riscv-tools bug.

closed 2018-10-01 by palmer-dabbelt

ferdous313, 2018-10-09:
ok Thanks!