riscv-software-src/riscv-tools #253: Assertion failed: !(eh.e_flags & EF_RISCV_RVC)

ferdous313, 2018-10-11:
Hi,
I am running a hello.c program in the emulator using 
 riscv64-unknown-elf-gcc -O2 -o hello hello.c
and then run it as, spike pk hello. It is giving the following error.
mada0:~/boom-template/rocket-chip/emulator$spike pk hello
Error:
../pk/elf.c:46: assertion failed: !(eh.e_flags & EF_RISCV_RVC)
Any suggestion will be appreciated.

aswaterman, 2018-10-11:
This results from running a binary with the C extension on a proxy kernel compiled without the C extension. Beyond that, I can’t help.

palmer-dabbelt, 2018-10-11:
The specific assumption here is that you've compiled PK to exactly match your hardware's ISA, so as a result if PK is compiled without the C extension then it should disallow loading binaries that use the C extension as they'll quickly emit an illegal instruction.  This isn't strictly necessary: a PK compiled without the C extension running on hardware with the C extension can run user code compiled for the C extension.  That said, it's a good way to keep users from shooting themselves in the foot because the assertion is a lot easier to debug than illegal instructions.

IIRC BOOM still doesn't support C, so what you really want to do here is compiler your application without the C extension.  You can do so by passing the `-march` argument to GCC, which controls the ISA that code will be generated for.

This isn't a riscv-tools bug, so I'm going to close the issue.  I wouldn't be opposed to improving the error message, feel free to open a PR on riscv-pk if you're interested.

closed 2018-10-11 by palmer-dabbelt

aswaterman, 2018-10-11:
FYI @palmer-dabbelt, PK does need to know about C to host C binaries, because it must emulate some C instructions, like misaligned loads and stores.