riscv-software-src/riscv-tools #254: assertion failed: !(eh.e_flags & EF_RISCV_RVC)

ferdous313, 2018-10-11:
I am trying to run hello in spike.
It is giving this error:

mada0:~/boom-template/rocket-chip/riscv-tools$riscv64-unknown-elf-gcc -o hello hello.c 
mada0:~/boom-template/rocket-chip/riscv-tools$spike pk hello 
../pk/elf.c:46: assertion failed: !(eh.e_flags & EF_RISCV_RVC)

hello.c
#include <stdio.h>
int main(void) {
printf("Hello!\n");
return 0;
}
Please help me solve this. Any suggestion will be appreciated.

ferdous313, 2018-10-11:
Sorry I did not check the closed one earlier. Please close this link.

ferdous313, 2018-10-11:
@palmer-dabbelt 
I am not sure about the syntax of -march as this is not available so much in online.
I tried and gives this error:
riscv-tools$riscv64-unknown-elf-gcc  -march=rv64gc -mabi=lp64 -o hello11 hello 
hello: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status

Please suggest the syntax.

jim-wilson, 2018-10-11:
hello is probably a binary file.  A C/C++ compiler works better when you pass a C/C++ file to it.  You can use the file command to see what kind of file hello is.  Try typing "file hello".

ferdous313, 2018-10-11:
mada0:~/boom-template/rocket-chip/riscv-tools$file hello
hello: C source, ASCII text

jim-wilson, 2018-10-11:
The compiler assumes that files that end with a .c are C source files, and that files with no ending are linker input files.  If you are using a different naming scheme, then you have to tell the compiler that it is a C file.  You can use the -x <language> option for that, where language can be c, c++, ada, f77, f95, etc.  So try using "riscv64-unknown-elf-gcc -march=rv64gc -mabi=lp64 -o hello11 -x c hello".  But the simpler solution is "mv hello hello.c" and then compile hello.c.