riscv-software-src/riscv-tools #258: Error when booting a Linux on riscv

Jeremygu, 2018-10-18:
Hi,

After building the linux-4.18.2 with patching riscv-linux-4.18, BusyBox1.26.2 and initramfs (all steps following to this [README](https://github.com/riscv/riscv-tools/tree/priv-1.10#linuxman)), I run the command `spike bbl vmlinux` to boot the linux but get stuck:

```
jeremy@j-XPS-8700:~/workspace/riscv/linux-4.18.2$ spike bbl vmlinux
OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
Linux version 4.18.0-14455-gdfeb52b (jeremy@j-XPS-8700) (gcc version 7.2.0 (GCC)) #2 SMP Wed Oct 17 15:12:08 PDT 2018
bootconsole [early0] enabled
Initial ramdisk at: 0x(____ptrval____) (541185 bytes)
Zone ranges:
  DMA32    empty
  Normal   [mem 0x0000000080200000-0x00000000ffffffff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000080200000-0x00000000ffffffff]
Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
software IO TLB [mem 0xfa3fe000-0xfe3fe000] (64MB) mapped at [(____ptrval____)-(____ptrval____)]
elf_hwcap is 0x112d
percpu: Embedded 17 pages/cpu @(____ptrval____) s29592 r8192 d31848 u69632
Built 1 zonelists, mobility grouping on.  Total pages: 516615
Kernel command line: 
Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
Sorting __ex_table...
Memory: 1988304K/2095104K available (5408K kernel code, 328K rwdata, 1731K rodata, 724K init, 810K bss, 106800K reserved, 0K cma-reserved)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Hierarchical RCU implementation.
	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
Console: colour dummy device 80x25
console [tty0] enabled
bootconsole [early0] disabled
```

Could you please give any advice? Appreciate it very much.

jim-wilson, 2018-10-18:
On Wed, Oct 17, 2018 at 10:01 PM Haifeng Gu <notifications@github.com>
wrote:

> Hi,
>
> After building the linux-4.18.2 with patching riscv-linux-4.18,
> BusyBox1.26.2 and initramfs (all steps following to this README
> <https://github.com/riscv/riscv-tools/tree/priv-1.10#linuxman>), I run
> the command spike bbl vmlinux to boot the linux but get stuck:
>

That README you cited if 4 years old and hopelessly out of date.  You
should not assume that anything in there is correct.  Because most of the
info in there is wrong.

Which exact linux are you using?  If you are using upstream linux from
kernel.org, I think 4.19 is the first one that is bootable.  The older ones
are missing critical drivers.  While 4.19 works, I think it only works on
qemu, not on the unleashed hardware, because drivers for the hardware are
still missing.  You may want to use riscv/riscv-linux instead until
everything we need is upstream.

There are many easier ways to get linux nowadays.  You can download debian,
fedora, opensuse, openembedded, etc.  There is also on github
sifive/freedom-u-sdk which can build a linux kernel with buildroot and boot
it.

> jeremy@j-XPS-8700:~/workspace/riscv/linux-4.18.2$ spike bbl vmlinux
>
>
Why are you using spike?  Spike is the golden reference, but doesn't
support any peripherals.  You should use qemu instead if you want to do any
useful work.

>
>
> Console: colour dummy device 80x25
> console [tty0] enabled
> bootconsole [early0] disabled
>
> Could you please give any advice? Appreciate it very much.
>

When using spike, this usually indicates that you are using the wrong
serial driver.  The new driver is better than the old driver, but does not
work on spike.  To get the old driver, you have to enable
CONFIG_HVC_RISCV_SBI.

Jim

Jeremygu, 2018-10-18:
Hi Jim,

Thanks for your response. I use the linux-4.18.2 downloaded from kernel.org since riscv-linux has the 4.18 version.
I am new to this, and may not need many peripherals now for getting start. Therefore, I followed the instructions tring to boot a small linux by using spike.
As you mentioned CONFIG_HVC_RISCV_SBI, I checked that it had been set `y` in the `linux-4.18.2/.config` file. Then I rebuilt `vmlinux`, but got the same problem..

BTW, I fully agree with you that there're many linux that we can get to use. Just now I tried the freedom-u-sdk by simply `unset RISCV` and `make sim`. Very easy to boot it up.
But I really want to figure out what the problem is there.

jim-wilson, 2018-10-18:
We are still in the process of upstreaming RISC-V linux kernel support.  I believe 4.19 is the first one from kernel.org that will work, but only on qemu and not on unleashed hardware yet (don't know about spike).  riscv-linux is kernel.org plus all of the missing patches that haven't been accepted upstream yet.  So 4.18 from riscv-linux may work even if 4.18 from kernel.org does not.  If you want more info about RISC-V linux, you should probably ask on the linux-riscv mailing list instead of here.