riscv-software-src/riscv-tools #263: gcc not finding libraries that are under glibc.

AhmedSamara, 2018-11-26:
(simplified version for brevity)
Did you figure out what exactly was causing this?

I'm having a similar problem of not being able to include .h files that are present.
```
#include <execinfo>
 
 int main()
 {
     return 0;
 }
```
```
ex.c:1:20: fatal error: execinfo: No such file or directory
 #include <execinfo>
```

execinfo does exist under `/riscv-gnu-toolchain/riscv-glibc/include/execinfo.h`

But `riscv64-unknown-elf-gcc` doesn't seem to be aware of it.

jim-wilson, 2018-11-26:
That isn't valid C code, and I don't think it is valid C++ code either.  Use instead
```
#include <execinfo.h>
```
to include the execinfo.h file.

AhmedSamara, 2018-11-29:
My mistake, I re-typed it to have a bare-metal program but I get the same error with the correct syntax.
```
#include <execinfo.h>

int main()
{
	return 0;
}
```

`ex.c:1:10: fatal error: execinfo.h: No such file or directory
 #include <execinfo.h>
```

I looked in the `riscv-gnu-toolchain/build` folder and there's nothing about glibc so I think it's not being built at all. 

I tried building it manually (instead of using `build.sh`) I got this error when running

```
export glibc_install="/path/to/riscv-glibc/build/install"
 ../configure --prefix "$glibc_install"
```

I get this error
```
checking for sysdeps preconfigure fragments... aarch64 alpha arm hppa i386 m68k microblaze mips nios2 powerpc riscv Unable to determine XLEN
```

Not sure how to set XLEN

jim-wilson, 2018-11-29:
riscv64-unknown-elf-gcc is a cross compiler for embedded targets using the newlib C library.  Newlib does not have an execinfo.h file.

If you want a linux compiler, use --enable-linux when configuring riscv-gnu-toolchain.  --disable-linux is the default.

Building glibc is complicated.  Don't try to build it manually.  Just configure for a linux target instead.

AhmedSamara, 2018-11-29:
Did I do this right? I added `--enable-linux` to `build.sh`

`build_project riscv-gnu-toolchain --prefix=$RISCV --enable-linux` and it appears to have worked because I can see all of the build files in `riscv-gnu-toolchain/build`, but I still can't compile the program above.

jim-wilson, 2018-11-30:
Are you using riscv64-unknown-linux-gcc?  What exact command are you typing and what exact error are you getting?

Yes, that configure command for riscv-gnu-toolchain looks right.  In general, I recommend using riscv-gnu-toolchain instead of riscv-tools, as riscv-tools is difficult to build, and has an old obsolete and known broken version of the toolchain.

AhmedSamara, 2018-11-30:
Compiled the toolchain with [this script](https://github.com/riscv/riscv-tools/blob/d4f3490ed2bb54bcc7be7b2a11f764cafce579fb/build.sh#L23), but with `--enable-linux` appended to line 32.

Then executed this command:
```
>riscv64-unknown-elf-gcc ex.c 
ex.c:1:10: fatal error: execinfo.h: No such file or directory
 #include <execinfo.h>
          ^~~~~~~~~~~~
compilation terminated.
```

jim-wilson, 2018-11-30:
You typed riscv64-unknown-elf-gcc.  As explained before, this won't work, because the elf compiler is for embedded development using the newlib C library, which does not support the execinfo.h file.

You need to use riscv64-unknown-linux-gnu-gcc instead.

Note that embedded elf binaries and linux binaries are not interchangeable.  Linux binaries only work on linux system.  Embedded elf binaries only work on embedded elf systems.  You need to decide what your target is, embedded elf or linux, and then program accordingly.

AhmedSamara, 2018-11-30:
Ah that makes sense. Thank you very much for your help and patience in explaining this, I appreciate it a lot.

closed 2018-11-30 by AhmedSamara