riscv-software-src/riscv-tools #264: mac os

eric86103, 2018-11-28:
I'm trying to install toolchain on macOS mojave 10.14.1
and followed the steps in readme and run
```
CC=gcc-5 CXX=g++-5 ./build.sh
```
(my gcc version is 4.2.1 so I think I have to add the first two arguments?)
However, I can't build the toolchain, and got some seeming error when configure riscv-openocd
```
Configuring project riscv-openocd
../../jimtcl/configure.gnu: line 1: -e: command not found
../../../../../src/jtag/drivers/libjaylink/configure.gnu: line 1: -e: command not found
```
Reach for help, thanks!

eric86103, 2018-11-28:
and when configure riscv-gnu-toolchains, I got the following error... and failed the building process.
`../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In function 'void ivs_params_clear(ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:98:29: error: 'isl_id_free' was not declared in this scope
       isl_id_free (it->first);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'tree_node* translate_isl_ast_to_gimple::gcc_expression_from_isl_ast_expr_id(tree, isl_ast_expr*, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:293:26: error: 'isl_id_free' was not declared in this scope
   isl_id_free (tmp_isl_id);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'tree_node* translate_isl_ast_to_gimple::gcc_expression_from_isl_ast_expr_id(tree, isl_ast_expr*, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:293:26: error: 'isl_id_free' was not declared in this scope
   isl_id_free (tmp_isl_id);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'edge_def* translate_isl_ast_to_gimple::translate_isl_ast_for_loop(loop_p, isl_ast_node*, edge, tree, tree, tree, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:669:72: error: 'isl_id_get_user' was not declared in this scope
       ast_build_info *for_info = (ast_build_info *) isl_id_get_user (id);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:672:22: error: 'isl_id_free' was not declared in this scope
       isl_id_free (id);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function 'edge_def* translate_isl_ast_to_gimple::translate_isl_ast_node_user(isl_ast_node*, edge, ivs_params&)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:876:55: error: 'isl_id_get_user' was not declared in this scope
   poly_bb_p pbb = (poly_bb_p) isl_id_get_user (name_id);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:882:23: error: 'isl_id_free' was not declared in this scope
   isl_id_free (name_id);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In function 'isl_id* ast_build_before_for(isl_ast_build*, void*)':
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2886:61: error: 'isl_space_dim' was not declared in this scope
   int dimension = isl_space_dim (schedule_space, isl_dim_out);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2890:33: error: 'isl_space_free' was not declared in this scope
   isl_space_free (schedule_space);
../../../riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2891:73: error: 'isl_id_alloc' was not declared in this scope
   isl_id *id = isl_id_alloc (isl_ast_build_get_ctx (build), "", for_info);
make[2]: *** [graphite-isl-ast-to-gimple.o] Error 1
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
`
this is my gcc-5 -v
`Using built-in specs.
COLLECT_GCC=gcc-5
COLLECT_LTO_WRAPPER=/usr/local/Cellar/gcc@5/5.5.0_2/libexec/gcc/x86_64-apple-darwin14.5.0/5.5.0/lto-wrapper
Target: x86_64-apple-darwin14.5.0
Configured with: ../configure --build=x86_64-apple-darwin14.5.0 --prefix=/usr/local/Cellar/gcc@5/5.5.0_2 --libdir=/usr/local/Cellar/gcc@5/5.5.0_2/lib/gcc/5 --enable-languages=c,c++,objc,obj-c++,fortran --program-suffix=-5 --with-gmp=/usr/local/opt/gmp --with-mpfr=/usr/local/opt/mpfr --with-mpc=/usr/local/opt/libmpc --with-system-zlib --enable-libstdcxx-time=yes --enable-stage1-checking --enable-checking=release --enable-lto --enable-plugin --disable-werror --with-pkgversion='Homebrew GCC 5.5.0_2' --with-bugurl=https://github.com/Homebrew/homebrew-core/issues --disable-nls --enable-multilib
Thread model: posix
gcc version 5.5.0 (Homebrew GCC 5.5.0_2) 
`
g++-5 -v
`Using built-in specs.
COLLECT_GCC=g++-5
COLLECT_LTO_WRAPPER=/usr/local/Cellar/gcc@5/5.5.0_2/libexec/gcc/x86_64-apple-darwin14.5.0/5.5.0/lto-wrapper
Target: x86_64-apple-darwin14.5.0
Configured with: ../configure --build=x86_64-apple-darwin14.5.0 --prefix=/usr/local/Cellar/gcc@5/5.5.0_2 --libdir=/usr/local/Cellar/gcc@5/5.5.0_2/lib/gcc/5 --enable-languages=c,c++,objc,obj-c++,fortran --program-suffix=-5 --with-gmp=/usr/local/opt/gmp --with-mpfr=/usr/local/opt/mpfr --with-mpc=/usr/local/opt/libmpc --with-system-zlib --enable-libstdcxx-time=yes --enable-stage1-checking --enable-checking=release --enable-lto --enable-plugin --disable-werror --with-pkgversion='Homebrew GCC 5.5.0_2' --with-bugurl=https://github.com/Homebrew/homebrew-core/issues --disable-nls --enable-multilib
Thread model: posix
gcc version 5.5.0 (Homebrew GCC 5.5.0_2) 
`

jim-wilson, 2018-11-28:
If you just want a toolchain, don't use riscv-tools.  It is poorly maintained, difficult to build, and has an old obsolete and known broken version of the toolchain.  Use riscv-gnu-toolchain instead.

I don't have a mac, so can't comment on the mac specific errors.  The isl_id_free/etc problem is a versioning problem with isl and gcc.  See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=86724 for more info and the fix.

eric86103, 2018-11-28:
what do you mean "don't use ricv-tools", does not it include the riscv-gnu-toolchains?
And what I want eventually is to build a custom riscv-tools (with simulator, compiler, and etc.)

jim-wilson, 2018-11-28:
riscv-tools includes an old obsolete and known broken version of riscv-gnu-toolchain.  If you want a good version of riscv-gnu-toolchain, you need to clone and build riscv-gnu-toolchain.

Or if you really need other stuff from riscv-tools, and want to build them together, you can try doing something like "cd riscv-gnu-toolchain; git checkout master" which should give you the current version of riscv-gnu-toolchain.  Then avoid running a git submodule command at the top of the riscv-tools tree as that will give you the old obsolete and known broken version of riscv-gnu-toolchain again.  I don't know if this will work though.  Upgrading the toolchain may expose latent bugs in other parts of riscv-tools.

eric86103, 2018-11-28:
ok, first of all, thanks for the link provided, I solved the problems!
But we I type "cd riscv-gnu-toolchain; git checkout master"
`cd riscv-gnu-toolchain`
`git checkout master`
I tell me that my repo is up-todate, could you explain it more thoroughly?
Sincerely thanks!

jim-wilson, 2018-11-28:
riscv-tools uses git submodules, which hard codes submodules to specific commits.  The riscv-gnu-toolchain submodule is hard coded to an old commit that has old obsolete versions of stuff.  If you cd into riscv-gnu-toolchain and do a "git checkout master" it will switch that submodule to the current version.  You should have gotten interesting messages when you did this.  I get this
```
gamma05:2021$ git checkout master
warning: unable to rmdir riscv-binutils-gdb: Directory not empty
M	riscv-gcc
M	riscv-newlib
Previous HEAD position was 64879b2... Bump gdb
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
gamma05:2022$ 
```
So it did switch, and then tells me that I'm now up-to-date with the new commit.  Then you have to do a git submodule update in the riscv-gnu-toolchain dir to finish getting the rest of the toolchain.  And you can rm -rf the riscv-binutils-gdb dir which is no longer used.

If you are on the master branch, and use git log, you should see this as the first commit right now, though there may be another commit on top soon.
```
commit 75504a4f877ca5d1a7349fed30a1e5e4ed7eb606
Merge: d002dc7 f431e36
Author: Jim Wilson <jimw@sifive.com>
Date:   Tue Nov 27 11:02:38 2018 -0800

    Merge pull request #394 from riscv/configure-check
    
    Run config.status --recheck when configure changes.

```