riscv-software-src/riscv-tools #270: How to build on Centos 7

oadjemian, 2019-01-08:
Hi all,

I try to build on Centos 7.6 without success.

I update automake to version 1.15
# automake --version
automake (GNU automake) 1.15

I update autoconf to version 2.69
# autoconf --version
autoconf (GNU Autoconf) 2.69

I update libtool version 2.4.6
# libtool --version
libtool (GNU libtool) 2.4.6

 # ./build.sh
Starting RISC-V Toolchain build process

libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:42: installing 'build-aux/ar-lib'
configure.ac:37: installing 'build-aux/compile'
configure.ac:28: installing 'build-aux/config.guess'
configure.ac:28: installing 'build-aux/config.sub'
configure.ac:30: installing 'build-aux/install-sh'
configure.ac:30: installing 'build-aux/missing'
Makefile.am: installing './INSTALL'
libjaylink/Makefile.am: installing 'build-aux/depcomp'
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
libtoolize: 'AC_PROG_RANLIB' is rendered obsolete by 'LT_INIT'
configure.ac:12: error: possibly undefined macro: AC_MSG_WARN
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:206: error: possibly undefined macro: AC_DEFINE
configure.ac:610: error: possibly undefined macro: AC_MSG_NOTICE
autoreconf: /usr/local/bin/autoconf failed with exit status: 1

Thanks for your help,
Olivier

jim-wilson, 2019-01-08:
riscv-tools is difficult to build and unmaintained.  Don't use it.  If you want a toolchain, then use riscv-gnu-toolchain instead.

If you insist on wasting your time trying to build riscv-tools, then you are on the right track.  I think the part you are missing is that you need to grab some autoconf/automake patches from the upstream openocd tree in order to use these autoconf/automake versions.

The simple solution is to just install Ubuntu 16.04LTS on a machine, and then use that to build riscv-tools.

The even simpler solution is to just never try to build riscv-tools.

oadjemian, 2019-01-09:
Hi Jim,
I prefer to use CentOS/Redhat because most of all EDA tools are running under.
I have build all of the packages except openocd, I have used openocd CentOS bundle but it is not the same version that provided by riscv-tools. so what are autoconf/automake and other package version that are needed to be patched ?

jim-wilson, 2019-01-13:
It was over 6 months ago that I last looked at Centos support for a friend, and I can't find any of my instructions or build trees.  It looks like I didn't save anything, so I can't give you detailed instructions on how to do the build.

SiFive has releases that have been built on Centos.  Maybe you could use one of those?  See
https://www.sifive.com/boards
and scroll down to the prebuilt toolchain section.