riscv-software-src/riscv-tools #273: Error with compile c code

Mohsannaeem, 2019-02-06:
I  had installef risc-v toolchain in my pc linux as stated in risc_v website but now when i compiled my c code with risc-v-unknown.... Command .i face error No such file or directory .... Should i installed spike or any other   tools to compile my c code on my pc to create machine code/ assembly code

Mohsannaeem, 2019-02-07:
> You need to give a better bug report, including enough info to reproduce.
> What exact command are you running? What exact error are you getting? Did
> you verify that the program you are trying to run exists? If the program
> and input files exist, then you could be getting a ld.so error. Try running
> ldd on the binary that gives the error. If it says that ld.so is missing,
> then you have a binary that won't run on your OS version, in which case I
> need to know which linux distro you are running on your pc, and what
> version of it.

Thanks for your reply I just want to machine code/ assembly code  of my  c
code on linux in my pc and test my verilog code using this machine/
assembly code ....  I recently install risc v gnu toochain as described on
risc-v website ... But could not able to compile my c code ..... And can't
configure the write way  to  get assembly code ..... I am a beginner and
have not much insight about risc-v toolchain

Mohsannaeem, 2019-02-07:
> As mentioned before, I need enough info to reproduce. Which exact set of
> instructions on which risc-v web site did you use? There are many of them.
> I need a url. I also need to know exactly what command you typed, and
> exactly what error you got.

https://github.com/riscv/riscv-gnu-toolchain
I had followed these step and after this  i tryed to run  this error ... I
just delete all files and reinstall and follow  this website step again
with  this commands

And now after running  make report- linux i get  the error as showed in
second picture

Thanks for your reply

jim-wilson, 2019-02-07:
I don't see any pictures, so I still don't know what error you hit.  make report-linux should work, but it isn't a fatal error if it fails, because this is running the regression testsuite which runs many thousands of obscure tests.

If you are having trouble building your own compiler, you might try a compiler built by someone else.  If you are on a recent linux distro, like Ubuntu 18.04 or debian testing, you can just install a package to get a riscv64 linux compiler.  E.g. "apt install gcc-8-riscv64-linux-gnu" on my Ubuntu 18.04 system.  SiFive has some prebuilt toolchains available at https://www.sifive.com/boards