riscv-software-src/riscv-tools #277: How to remove the installed packages?

crumber, 2019-03-02:


jim-wilson, 2019-03-02:
Which installed packages do you want to remove?  OS packages?  What OS are you running?  Or riscv-tools itself?  If riscv-tools, where did you install it?  If you installed riscv-tools in its own dir, then you can just rm -rf it.

crumber, 2019-03-02:
I mean riscv-tools installed on Ubuntu

jim-wilson, 2019-03-02:
Did you try rm -rf for the riscv-tools install dir?  Assuming of course that you don't have anything else in there.

crumber, 2019-03-02:
And is this enough to remove all the toolchain? I didn’t try yet because the toolchain is installed in the directory /riscv/toolchain so I’m not sure that removing riscv-tools will remove all the toolchain

jim-wilson, 2019-03-02:
The toolchain only puts files in the install directory.  It does not install anything anywhere else.