riscv-software-src/riscv-tools #278: riscv32-unknow.elf-gcc: Command not found

maherca9, 2019-03-02:
I'm trying to follow this guide:
https://github.com/jerralph/riscv-vip-scr1-demo
and
https://github.com/jerralph/scr1/tree/riscv-vip-demo#prerequisites

But when I do "make run_modelsim BUS=AXI ARCH=I IPIC=0 CROSS_PREFIX=riscv32-unknown-elf-" I get this result:

https://i.ibb.co/S0GkqBC/Captura-de-pantalla-de-2019-03-02-15-18-26.png

I have also tried "make run_modelsim BUS=AXI ARCH=I IPIC=0 CROSS_PREFIX=riscv64-unknown-elf-" and this is the result:

https://i.ibb.co/9qT47NB/Captura-de-pantalla-de-2019-03-02-15-18-59.png

My .bashrc:
https://i.ibb.co/D8fgJtL/Captura-de-pantalla-de-2019-03-02-15-19-35.png

My folders:
https://i.ibb.co/gg690dC/Captura-de-pantalla-de-2019-03-02-15-19-56.png

My riscv-gnu-toolchain/bin:
https://i.ibb.co/5RFzKsB/Captura-de-pantalla-de-2019-03-02-15-20-19.png

aswaterman, 2019-03-02:
Assuming the installation succeeded, it sounds like you didn’t add
install-directory/bin to your PATH.

aswaterman, 2019-03-02:
Another possibility is that you built a riscv64-unknown-elf-gcc instead of
riscv32, so you could try that command instead.

jim-wilson, 2019-03-02:
This isn't a riscv-tools problem.  This is a jerralph/riscv-vip-scr1-demo problem.  You should try filing issues there.

>But when I do "make run_modelsim BUS=AXI ARCH=I IPIC=0 CROSS_PREFIX=riscv32-unknown-elf-" I get this result:

This fails because you don't have a riscv32-unknown-elf-gcc in your install dir.  Build one or download one from somewhere. 

>I have also tried "make run_modelsim BUS=AXI ARCH=I IPIC=0 CROSS_PREFIX=riscv64-unknown-elf-" and this is the result:

This fails because the toolchain binaries are linked against libraries that you don't have on your system.  Presumably you downloaded them from somewhere, and they aren't statically linked.  Either you can fix this by installing a package, or your downloaded toolchain binaries were compiled for a different OS and won't work on your system.  If the latter, then you need to talk to the people that you downloaded the toolchain binary from.  Or get a different one, by building one yourself or downloading from elsewhere.  If the former, then for the list of OS packages you need, see the riscv/riscv-gnu-toolchian README.md file.
    https://github.com/riscv/riscv-gnu-toolchain

And remember, riscv-tools is fubar, don't use it.  The toolchain is in riscv/riscv-gnu-toolchain, not riscv/riscv-tools.

maherca9, 2019-03-03:
> This fails because you don't have a riscv32-unknown-elf-gcc in your install dir. Build one or download one from somewhere.
> 
I don't know how to do it. I'm inexperienced with Linux and riscv. How can I get "riscv32-unknown-elf-gcc"? I realised I don't have it, but I didn't know how get it.
 
> This fails because the toolchain binaries are linked against libraries that you don't have on your system. Presumably you downloaded them from somewhere, and they aren't statically linked. Either [...]

I have also followed this guide, and I have downloaded those files. I used "Installation (Linux multilib)" for both 32-bit and 64-bit.

I suppose I am going to post my questions in "jerralph/riscv-vip-scr1-demo". But I have followed the instructions in riscv-tools and I didn't manage to run it.

maherca9, 2019-03-04:
I have solved the problem `make run_modelsim BUS=AXI ARCH=I IPIC=0 CROSS_PREFIX=riscv64-unknown-elf-` with `sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.6 /usr/lib/x86_64-linux-gnu/libmpfr.so.4`

However I want to know how get a riscv32-unknown-elf-gcc.

jim-wilson, 2019-03-04:
Forget about riscv-tools.  It is broken.  Use riscv/riscv-gnu-toolchain instead.  See the README.md file. 
 It builds a 64-bit toolchain by default.  When configuring, you can pass --with-arch=rv32gc --with-abi=ilp32d to get a 32-bit compiler.  Any supported combination of with-arch/with-abi should work.  The default is an embedded elf toolchain.  If you want a linux toolchain, add --enable-linux.  The default is to build for one arch/abi combination.  If you want a toolchain that supports multiple arch/abi combinations, add --enable-multilib.  There are about 13 multilibs that are built by default, which is not every possible combination but a reasonable set.

Normally, for an embedded elf target, you should never need a riscv32 compiler.  The convention is to build a riscv64 compiler, with multilib support, and generate 32-bit code by passing for instance -march=rv32gc -mabi=ilp32d.