riscv-software-src/riscv-tools #281: Now I need to port this set of tools to win10, may I ask through what way?I have tried mingw so far but failed. I wonder if there is another way

mouhuajun, 2019-03-14:
Now I need to port this set of tools to win10, may I ask through what way?I have tried mingw so far but failed. I wonder if there is another way

mouhuajun, 2019-03-14:
Or is there anyone doing something like that that we can talk to each other about

palmer-dabbelt, 2019-03-14:
Does windows 10 not support win32 applications?  I'd always assumed our binaries would just work there, though I don't know if I've ever tried.

jim-wilson, 2019-03-14:
Yes, windows 10 supports win32 applications.  The problem is that building linux stuff on windows is complicated if you've never done it before.

mingw+msys2 should work, but takes some time to learn how to use.

I find cygwin32 easier to use, but it has the downside that you need a big dll, and every minor version update creates a new incompatible dll, leading to problems managing the different dlls.  It is generally better to use mingw if you can.

Cross compiling from linux is even easier, but assumes you have a linux machine, and know how to cross compile.

An even easier way is to just use tools that someone else has built.  Liviu has gnu-mcu-eclipse.  SiFive has toolchains and an eclipse port on our web site.

I've never used WSL.

The fundamental problem here is not a RISC-V problem, it is a windows compilation problem.  You might try asking in a mingw help mailing list for instance.

mouhuajun, 2019-03-15:
> 是的，Windows 10支持win32应用程序。问题是如果你以前从未这样做过，在windows上构建linux的东西很复杂。
> 
> mingw + msys2应该可以工作，但需要一些时间来学习如何使用。
> 
> 我发现cygwin32更容易使用，但它有一个缺点，你需要一个大的dll，每个次要版本更新创建一个新的不兼容的DLL，导致管理不同的dll的问题。如果可以，通常最好使用mingw。
> 
> 从linux进行交叉编译更容易，但假设你有一台linux机器，并且知道如何交叉编译。
> 
> 更简单的方法是使用其他人构建的工具。Liviu有gnu-mcu-eclipse。SiFive在我们的网站上有工具链和日食端口。
> 
> 我从未使用过WSL。
> 
> 这里的根本问题不是RISC-V问题，而是windows编译问题。例如，您可以尝试在mingw帮助邮件列表中询问。
I have tried, but there will be many problems that the library does not exist, I do not know how to solve, I did not find a similar case on the Internet, gun-mcu-eclipse can only be connected to the development board to operate

closed 2019-03-15 by mouhuajun