riscv-software-src/riscv-tools #282: compile OpenOCD issue

guox9, 2019-03-23:
When I compile new riscv tool chain, next error is occurred. I didn't know what's the reason and cannot find any solution now.

Thanks

hostname:/rocket-chip/riscv-tools>./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:4506: warning: overriding commands for target `check-recursive'
Makefile:3917: warning: ignoring old commands for target `check-recursive'
Makefile:4506: warning: overriding commands for target `check-recursive'
Makefile:3917: warning: ignoring old commands for target `check-recursive'
Makefile:4506: warning: overriding commands for target `check-recursive'
Makefile:3917: warning: ignoring old commands for target `check-recursive'
../src/jtag/drivers/xds110.c: In function ‘usb_connect’:
../src/jtag/drivers/xds110.c:369:3: warning: implicit declaration of function ‘libusb_set_auto_detach_kernel_driver’ [-Wimplicit-function-declaration]
   (void)libusb_set_auto_detach_kernel_driver(dev, 1);
   ^
src/.libs/libopenocd.a(src_jtag_drivers_libocdjtagdrivers_la-xds110.o): In function `xds110_init':
xds110.c:(.text+0x1dea): undefined reference to `libusb_set_auto_detach_kernel_driver'
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/openocd] Error 1
gmake[1]: *** [all-recursive] Error 1
gmake: *** [all] Error 2

jim-wilson, 2019-03-23:
riscv-tools is not the toolchain.  This is a set of misc tools for hardware developers that has an old out-of-date and known broken copy of the toolchain.  If you want the gnu toolchain, use riscv/riscv-gnu-toolchain instead.  Don't use the toolchain inside riscv-tools for anything important.

The libusb_set_auto_detach_kernel_driver function should come from libusb-1.0.  On an x86_64 Ubuntu 16.04 linux system I see
gamma05:2057$ grep libusb_set_auto_detach_kernel_driver /usr/include/libusb-1.0/libusb.h
int LIBUSB_CALL libusb_set_auto_detach_kernel_driver(
gamma05:2057$ nm -D /usr/lib/x86_64-linux-gnu/libusb-1.0.so | grep libusb_set_auto_detach_kernel_driver
0000000000003f90 T libusb_set_auto_detach_kernel_driver
gamma05:2058$ 

So either you are missing this library, or  you have a too old version of it, or you have a too new version of it.

guox9, 2019-03-25:
Hi, @jim-wilson, Thanks for your detail guide of this issue. I checked the libusb version and it is 1.0.9. I tried to install other version libusb1 including libusb1-devel, but cannot resolve this problem. I didn't know why the libusb1 library is not correct (the OS is centos 6.10 and all package of libusb1 are installed) and will move to riscv-gnu-toolchain.

jim-wilson, 2019-03-25:
Centos 6 is a rather dated version of linux, though it is the EDA standard so it is pretty common.  Centos 6 may have a too old version of libusb to compile current riscv-openocd sources.  The SiFive toolchain includes its own copy of the libusb sources, perhaps to solve this problem.  This is sifive/freedom-tools.  You might need to do something similar to make this work on Centos 6.

You only need riscv-openocd if you want to use it to debug embedded code running on hardware.  There are other ways to do this, e.g. various commercial JTAG tools.  If you just want to use simulators, or have another jtag solution, then you don't need riscv-openocd.

guox9, 2019-03-28:
Hi, Jim-wilson,
Yes, you are right. I don't need openocd now. So I will close this issue and transfer to riscv-gnu-toolchain. Thanks you again for your reply that include much information valuable for me.
Best regards
Xiaoxu Guo

closed 2019-03-28 by guox9