riscv-software-src/riscv-tools #285: build.sh error about gcc

xuejiazidi, 2019-04-03:
Hi, 
I followed the Quickstart to install the risc-v tools.  However, I failed to run build.sh:

gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [file.o] Error 1
make: *** Waiting for unfinished jobs....
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [frontend.o] Error 1
make: *** [handlers.o] Error 1
make: *** [syscall.o] Error 1

I used Ubuntu 18.04.1 LTS.  My gcc version is 4.8.5.

Looking forward to you reply.

jim-wilson, 2019-04-03:
You need to use a RISC-V compiler to build RISC-V code.  You are trying to use an x86 compiler.

riscv-tools recently changed, and no longer includes a compiler.  You have to build that separately and put it on your path.  There needs to be a riscv64-unknown-elf-gcc binary on your PATH for the build to work.

Note that riscv-tools includes a few misc tools for hardware developers.  If what you really want is a toolchain, then riscv-tools is the wrong place to look.  The quick start guide points out a few ways to get a RISC-V compiler.  You can also try the github riscv/riscv-gnu-toolchain project for a compiler.

xuejiazidi, 2019-04-03:
I want to use the Rocket Chip Generator to generate instantiate the RISC-V Rocket Core.  Then simulate the core.  riscv-tools are involved in the Rocket Chip Generator Quick Instructions.  So I tried to install the riscv-tools.

So I need a RISC-V compiler to build the riscv-tools, is that correct?  Should I use riscv/riscv-gnu-toolchain project to build a compiler, then build the riscv-tools?

jim-wilson, 2019-04-03:
You can use any method to build the compiler.  crosstools-ng one such method.  riscv/riscv-gnu-toolchain is another.  crosstools-ng is mentioned in the docs before the quickstart section starts.

closed 2019-04-04 by xuejiazidi