riscv-software-src/riscv-tools #287: How could i install riscv32-unknown-elf to generate elf file for 32bit  processor...

Mohsannaeem, 2019-04-08:
I had already installed riscv-gnu-toolchain and  could able to use riscv64-unknown-elf to generate the 64 bit elf file but the problem is i want 32 bit elf file and for this I need riscv32-unknown-elf..How could i do this ....
[image]

jim-wilson, 2019-04-08:
riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 will generate code for a simple RV32 core without an FPU.  You can use any valid combination of arch and abi here, to generate 32 or 64-bit code, with or without FP, etc.

Mohsannaeem, 2019-04-09:
When i add -march=rv32imac -mabi=ilp32 it give error 
[image]

closed 2019-04-09 by Mohsannaeem

jim-wilson, 2019-04-09:
Did you build the toolchain yourself?  If so, then you need to add --enable-multilib when configuring to get a toolchain that supports multiple arches and ABIs.

Or use a pre-built toolchain that was configued with multilib support that you can get from various places.  SiFive for instance has pre-built toolchains on our web site, on the boards and software page.

Or if you want to build your own 32-bit toolchain that only supports one arch/abi, then you can use --with-arch=rv32imac --with-abi=ilp32 when configuring the toolchain.  As before, any valid arch and abi combination should work.

Mohsannaeem, 2019-04-09:
I rerun these command
$ sudo ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
$sudo make linux
$./configure --prefix=/opt/riscv --enable-multilib
$make linux

But While compiling with -march=rv32imac -mabi=ilp32
I got same error as before picture

jim-wilson, 2019-04-09:
This command
  ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
will give you a toolchain that only has one copy of the libraries compiled for -march=rv32gc/-mabi=ilp32d.  If you then compile with -march=rv32imac/-mabi=ilp32 you will get errors, because your code won't be abi compatible with the libraries.

You can fix this by using --with-arch=rv32imac --with-abi=ilp32, which gives a compiler that has only -march=rv32imac/-mabi=ilp32 compiled libraries and will only work with those -march/-mabi options.

Or you can fix this by adding --enable-multilib, which will build about 14 copies of the libraries for various march/mabi option combinations.  It doesn't cover every possible choice, but it covers most of the common ones.  Obviously, this toolchain will take longer to build, and will use more disk space in the build and install trees, because it has more copies of the libraries.  This will let you have one compiler that works with multiple arch/abi choices though.

The normal scheme is to build a multilib 64-bit compiler.  Since 64-bit is the default, you do not need any --with-arch or --with-abi option.  You only need the --enable-multilib option.  This gives you a compiler that can generate 32-bit or 64-bit code for various combinatins of -march and -mabi.

Mohsannaeem, 2019-04-09:
Thanks for Your reply ..Its very helpfull for me ...
Now the problem is While make linux after ./config..... got this error
[image]

And my log file is 
[config.log](https://github.com/riscv/riscv-tools/files/3061117/config.log)

jim-wilson, 2019-04-09:
There is a config.log file in each subdirectory that has a configure script.  You attached the top level one, which is not the one that failed.  However, it has your configure options
   ./configure --prefix=/opt/riscv --with-arch=rv32imac --with-abi=ilp32d
That won't work, as you have requested a non-fp arch with a fp abi.  You can't use a ilp32d ABI unless you have a 'd' or 'g' in the arch.  If you want a non-fp arch, then you want --with-abi=ilp32 which is the soft-float ABI.

Mohsannaeem, 2019-04-10:
Thanks for your reply but i get this warni while configure
[image]

And Then make file give same error as previous
https://github.com/riscv/riscv-tools/issues/287#issuecomment-481452884  
...What should i do

jim-wilson, 2019-04-11:
It looks like you have a very old and obsolete copy of riscv-gnu-toolchain.  Where did you get your riscv-gnu-toolchain sources?  You shoudl git clone riscv/riscv-gnu-toolchain to get a good copy of it.

Mohsannaeem, 2019-04-11:
Thanks for Your reply .....Now i am just compiling new riscv-gnu-toolchain  and i followed steps as 
https://github.com/riscv/riscv-gnu-toolchain
everything going perfect  ..But when i run 
$./configure --prefix=/opt/riscv --enable-multilib
after this $make sudo
 give error as shown in following picture
[image]

jim-wilson, 2019-04-11:
You typed "make linux".  I don't recommend doing that.  If you want a linux toolchain, then I recommend using --enable-linux when configuring, and building the embedded elf and linux toolchains separately.  But you said you wanted an embedded elf compiler at the start, so I don't know why you are using make linux.

Multilibbing is common for embedded elf toolchains, but not common for linux toolchains.  We might not have tested this combination recently and it may not be working.  I'd have to check.  If you really do need a linux toolchain, you probably don't need to multilib it.

Anyways, if you want to know why the build failed, you should examine the config.log file in the 32-bit glibc build dir to see what is wrong.

In general, I would not recommend doing everything as root via sudo.  You might try building and installing into a dir that you have write access to.  Or just give yourself write access to /opt/riscv so that you don't need sudo for the build.

Mohsannaeem, 2019-04-11:
Exactly i need toolchain for embedded  ..Then What should i do  to configure multilibbing  ...Honestly I write make linux because it is written on Readme of Riscv_gnu-toolchain ..
Please tell me What should i do to enable multilib

jim-wilson, 2019-04-11:
An embedded elf toolchain is the default.  Just configure with --enable-multilib, and then type "make" to build it.

Mohsannaeem, 2019-04-11:
One more question i had run make Linux after  
$./configure --prefix=/opt/riscv --with-arch=rv32imac --with-abi=ilp32
 Did I  is alright .... And can you please suggest me any reference to understant  varient of riscv-gnu-toolchain ....

jim-wilson, 2019-04-11:
The general process for building a multilib embedded elf toolchain is
```
mkdir X-build
cd X-build
../riscv-gnu-toolchain/configure --enable-multilib --prefix=/opt/riscv
make
```

The easiest way to get an embedded elf toolchain is to download a prebuilt one that someone else built, e.g. like the ones on the SiFive web site.

Palmer Dabbelt has a blog series that talks about the toolchain
    https://www.sifive.com/blog/all-aboard-part-0-introduction

myw2s-git, 2019-10-15:
Hi

I downloaded the pre-built toolchain from syntacore git repository. You can see the config.:

osboxes@osboxes:~/RFS/riscv/toolchain/sc-riscv64-unknown-elf-gcc-20180126-linux64/riscv-unknown-elf-gcc/bin$ ./riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/osboxes/RFS/riscv/toolchain/sc-riscv64-unknown-elf-gcc-20180126-linux64/riscv-unknown-elf-gcc/bin/../libexec/gcc/riscv64-unknown-elf/7.1.1/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/scrtools/work/host_work/tools/upstream-7.1-rv32e/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/scrtools/180115/ge5275d6f --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-headers=/opt/scrtools/180115/ge5275d6f/riscv64-unknown-elf/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 7.1.1 20170509 (GCC) 

I wrote a simple printf program to show the "hello! RISC-V". 

1. I compile it by "riscv64-unknown-elf-gcc -o hello.o hello.c". 
2. Run "spike pk hello.o"
3. It shows: 
bbl loader
hello! RISC-V

Suppose the hello program use default option "march=rv64imafdc", "mabi=lp64d ". Because I will use RV32IMC, therefore, I compile the program with the following option:

"riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -o hello.o hello.c"

Then run the spike again with: "spike pk hello.o"
bbl loader
**_../pk/elf.c:40: assertion failed: IS_ELF64(eh)_**

Why does it assert? Do I need to specify the ISA is RV32IMAC?? like "spike --isa=RV32IMAC pk hello.o"?? I tried but nothing printed...I wonder the problem is in compilation or the incorrect argument used when running spike.

Jason

jim-wilson, 2019-10-15:
New issues for new questions please.  Yes you have to pass an isa option to spike.  I would guess that the 32-bit support in pk doesn't work.  I don't know if this has ever worked.  I see that you are using an old toolchain with old glibc.  There could be versioning problems between glibc and pk if they are using different ABI versions.  You can try to debug and fix pk.  There is no one maintaining pk so it is unlikely anyone will fix this.  Or you can use a recent toolchain from riscv/riscv-gnu-toolchain along with user mode qemu.  This is known to work for both 64-bit and 32-bit.