riscv-software-src/riscv-tools #289: How to make riscv-gcc compile new csr register？

245950258, 2019-05-20:
I am working on rocket-chip. Recently, I added a new CSR register to my rocket-chip. However, it looks like that the RISCV-GCC can't compile my program.
    So, I want to know if the RISCV-TOOLS supports to add a new CSR register.

jim-wilson, 2019-05-20:
The toolchain is riscv-gnu-toolchain, not riscv-tools.

You can always use the CSR register number.  If you want to use the CSR name, then you need to modify binutils to add assembler/disassembler support for the new name.  In binutils, see the file include/opcode/riscv-opc.h and look for example at the CSR_MSTATUS support.  You need two lines, one which gives the register number, and one which maps the register number to the name string.

245950258, 2019-05-23:
Thank you for helping me.

closed 2019-05-23 by 245950258