riscv-software-src/riscv-tools #290: compile tools error

altanary, 2019-05-31:
when I do make, the issue occurs as following;

"
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv64imc -mabi=lp64 -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mabi=lp64’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make: *** [file.o] Error 1
"

FYI, my gcc version is "gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.4)"

Hope for some advice.

jim-wilson, 2019-05-31:
You are trying to use an x86 compiler on RISC-V code.  That won't work.  You have to use a RISC-V compiler.  riscv-tools does not include a compiler.  It is a collection of misc tools for hardware designers.  The toolchain is in riscv-gnu-toolchain.  Or alternatively, you can use crosstool-ng to build a compiler.  The README.md file suggests using crosstool-ng.  You can find example commands for building a crosstool-ng compiler in the .travis.yml file.

Crosshairs, 2019-08-24:
> You are trying to use an x86 compiler on RISC-V code. That won't work. You have to use a RISC-V compiler. riscv-tools does not include a compiler. It is a collection of misc tools for hardware [...]

I have intalled the riscv-gnu-toolchain.But  gcc: error: unrecognized argument in option ‘-mcmodel=medany’.

jim-wilson, 2019-08-24:
You didn't give us much info to work on.  I'm assuming that it was riscv-pk that failed to build.  You mentioned that you did install a riscv compiler.  Maybe you installed the wrong one?  You need a riscv64-unknown-elf-gcc.  If you look at the build.sh script, it has
```
CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-elf
```
so configure will look for a riscv64-unknown-elf-gcc and if it finds it then it will put in riscv-pk/build/Makefile the line
```
CC            := riscv64-unknown-elf-gcc
```
If configure does not find riscv64-unknown-elf-gcc then it will use gcc, and the build will fail with an error complaining about the -mcmodel option.

scotws, 2019-09-27:
This is also a problem if you built the toolchain with `riscv64-unknown-linux-gnu-gcc` - it will fail the same way. You have to change line 22 to `-host=riscv64-unknown-linux-gnu`. It might be worth changing this part of the script to check if `risc64-unkown-elf` is even present in `$RISCV/bin`, and if not, either throw a useful error or use what it finds. 

Also, even if you change line 22 in this scenario, the script still breaks in line 23 for the risc-tests because of `--prefix=$RISCV/riscv64-unknown-elf`. Unfortunately, changing this to `$RISCV/risc64-unknown-linux-gun` doesn't fix the problem, and even splitting it up in `--prefix` and `--host` entries as in line 22 doesn't work -- somewhere, that `risc64-unknown-elf` seems to be hard-coded. 

Maybe it's me, but as somebody who is trying to install the system from scratch for the first time, the whole toolchain/tools split seems terribly confusing. Part of it is the naming, the other part is that I am getting the impression that the compiler stuff was once part of tools as well, and so half of the documentation on the internet seems to just talk about tools?

jim-wilson, 2019-09-27:
riscv-tools isn't actively maintained.  Until someone volunteers to start maintaining it, it will remain a mess.  It is primarily tools for hardware developers.  If you aren't developing your own hardware, then you don't need it, and it is probably best to avoid it.

There is no intention that this should ever work with a linux compiler, and you should not be trying to fix it to work with a linux compiler.  You need an embedded elf compiler.

Yes, the toolchain used to be a git submodule, but it was an old, obsolete, and broken version of the toolchain, which due to the lack of maintenance never got updated to a working toolchain.  Since riscv-tools doesn't stress the compiler, it didn't matter that the compiler was broken.  But the fact that the broken compiler was there resulted in a lot of annoying useless bug reports against the toolchain, so I complained repeatedly, and this was eventually fixed by just removing the toolchain submodule.  

I've suggested renaming riscv-tools a few times to try to reduce the confusion, but nothing ever came of that.

CodeSwimBikeRunner, 2019-10-27:
How are we supposed to compile riscv assembly?

jim-wilson, 2019-10-27:
You can build a compiler from riscv/riscv-gnu-toolchain.  You can use crosstool-ng to build a compiler.  You can install a system package to get a cross compiler if you have a new enough linux distro.  You can download a pre-built compiler from a number of places, such as the SiFive web site.  Etc.

scotws, 2019-10-27:
@CodeSwimBikeRunner To expand on what @jim-wilson wrote: You can use the GCC compiler suite in various pre-built versions the write the assembler code in a file that ends with .s and assemble that. If you're new to the GCC assembler, there will be cursing involved: There doesn't seem to be a newbie friendly introduction, it works on so many hardware platforms that the options are endless, and like so much of the Unix world it suffers from Stupid Dropped Letter Syndrom (".globl" instead of ".global", just like "umount" and "creat()"). It's a pain to use directly, but it is insanely powerful. At some point, the RISC-V world will develop enough that somebody will write a useful introduction to it.
 
Until then, if you just want to get your feet wet, Anthony Dos Reis has just published an intro to RISC-V assembly at https://www.amazon.com/RISC-V-Assembly-Language-Anthony-Reis/dp/1088462006 that has its own small assembler. It's nowhere near has powerful as the GCC, but far easier to understand for beginners and there is no complicated tool chain to set up. 

What I ended up doing, BTW, is Fedora RISC-V with QEMU (https://github.com/scotws/RISC-V-tests/blob/master/docs/qemu_fedora_riscv_setup.md). This gives you a complete Linux system with all necessary libraries and tools. What would be nice of course is if the RISC-V foundation were to publish an official Docker container with QEMU and Fedora (or whatever) pre-built so you could just download the thing, ssh into it and get started. If such a container exists, I'm unaware of it.

jacobyugit, 2019-11-09:
I have successfully got rid of this: "error: unrecognized argument in option ‘-mcmodel=medany’ " after installation of "riscv/riscv-gnu-toolchain" as per the suggestion of jim-wilson.  Thank a lot!

muhammadsalmanafzal, 2020-12-14:
What @jim-wilson said is **maybe** true, you need to compile all the components separately. For the error which is due to riscv-pk, I was able to remove it and compile it perfectly by following the follow steps:

1. Compile **64-bit** toolchain from [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain) repo.
2. Add gcc to you **PATH** by adding following lines to your **~/.\<shell\>rc**
```
export example1=/path/to/bin/of/toolchain
export example2=/path/to/toolchain
export PATH="$example1:$PATH"
export PATH="$example2:$PATH"
```
3. Restart bash.
4. Now try to build it again by following instructions in the standard [riscv-pk](https://github.com/riscv/riscv-pk) repo.

Hope this helps. If you have any more issues, do reach out. Thanks.

MikeOpenHWGroup, 2021-02-18:
FWIW:
> You can download a pre-built compiler from a number of places, such as the SiFive web site. Etc.

I did just that.  GoTo: https://www.sifive.com/software and you're done.   Took 2 minutes, but I'll never get back the preceeding 2 hours.  Sigh.

Thanks @jim-wilson