riscv-software-src/riscv-tools #292: error in runing ./bulid.sh

JunnanLi, 2019-07-11:
Hi, I meet an error in runing "./bulid.sh"

Building project riscv-isa-sim
../spike_main/disasm.cc: In constructor ‘disassembler_t::disassembler_t(int)’:
../spike_main/disasm.cc:275:1: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 disassembler_t::disassembler_t(int xlen)
 ^~~~~~~~~~~~~~
Installing project riscv-isa-sim
mkdir /home/lijunnan/code/riscv-tools/riscv-tools/include
mkdir /home/lijunnan/code/riscv-tools/riscv-tools/include/fesvr
mkdir /home/lijunnan/code/riscv-tools/riscv-tools/lib
mkdir /home/lijunnan/code/riscv-tools/riscv-tools/lib/pkgconfig

Removing existing riscv-pk/build directory
Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [file.o] Error 1

jim-wilson, 2019-07-11:
You need a riscv64-unknown-elf-gcc cross compiler before you can build riscv-tools.  You can build one from riscv/riscv-gnu-toolchain, or from crosstool-ng if you enable experimental support, or from FSF sources if you know how to do that.  Or you can download a pre-built one from various places such as sifive.com/boards.

JunnanLi, 2019-07-11:
@jim-wilson 
Hi,
I have installed "riscv64-unknow-elf-gcc compiler" (according to the readme of riscv/riscv-gnu-toolchain) before building riscv-tools. Do I need to export some commands?

from Junnan Li

jim-wilson, 2019-07-12:
build.common has
    PATH="$RISCV/bin:$PATH"
So you must set RISCV to point at the prefix where the compiler was installed such that $RISCV/bin/riscv64-unknown-elf-gcc exists and is a working cross compiler.

snat99-hub, 2019-10-22:
Another error from build.sh. What a bloody mess, leads to colossal waste of time. Appreciate Jim's lone-soldier approach in supporting this collegiate toolset, but it is not in publically usable form, to put it kindly. Surprising, given  that  RiscV seems to getting promoted widely.

On the other hand I found a very clean, cool, lightweight open-source toolset RISCV OVPSIM from Imperas Software. Builds and runs examples from hello_world to linpack on an ISA simulator right off the bat, with clean trace output. Doesn't need all this baggage of full system simulation infrastructure and gcc cross compiler etc,  just to get started. Comes in Windows and Linux forms.  Good stuff. I recommend people switching to using that.

zhuzhizhan, 2019-12-03:
I meet the same problem with ubuntu 18. But I have compiled successfully with ubuntu 16.

zhuzhizhan, 2019-12-04:
> build.common has
> PATH="$RISCV/bin:$PATH"
> So you must set RISCV to point at the prefix where the compiler was installed such that $RISCV/bin/riscv64-unknown-elf-gcc exists and is a working cross compiler.

So we must build riscv-gnu-toolchain first?

jim-wilson, 2019-12-05:
Yes, you must build riscv-gnu-toolchain first, and it must be a riscv64-unknown-elf toolchain.  Or you might be able to download a pre-built one.  There are some prebuilt toolchains at sifive.com/boards
for instance.

zhuzhizhan, 2019-12-06:
> Yes, you must build riscv-gnu-toolchain first, and it must be a riscv64-unknown-elf toolchain. Or you might be able to download a pre-built one. There are some prebuilt toolchains at sifive.com/boards
> for instance.

Got it, thanks.

0ena, 2020-07-31:
Hi @jim-wilson ,

I am facing the same issue as the OP. I would like to give it a try with the gnu-toolchain before going to a prebuilt toolchain.
I have followed the instructions for installing the https://github.com/riscv/riscv-gnu-toolchain, more specifically I installed the Linux multilib, using the following commands:

```
./configure --prefix=/opt/riscv --enable-multilib
make linux
```

This created a ./bin folder inside the installation folder that contains the following:
```
riscv64-unknown-linux-gnu-addr2line
riscv64-unknown-linux-gnu-ar
riscv64-unknown-linux-gnu-as
riscv64-unknown-linux-gnu-c++
riscv64-unknown-linux-gnu-c++filt
riscv64-unknown-linux-gnu-cpp
riscv64-unknown-linux-gnu-elfedit
riscv64-unknown-linux-gnu-g++
riscv64-unknown-linux-gnu-gcc
riscv64-unknown-linux-gnu-gcc-10.1.0
riscv64-unknown-linux-gnu-gcc-ar
riscv64-unknown-linux-gnu-gcc-nm
riscv64-unknown-linux-gnu-gcc-ranlib
riscv64-unknown-linux-gnu-gcov
riscv64-unknown-linux-gnu-gcov-dump
riscv64-unknown-linux-gnu-gcov-tool
riscv64-unknown-linux-gnu-gdb
riscv64-unknown-linux-gnu-gdb-add-index
riscv64-unknown-linux-gnu-gfortran
riscv64-unknown-linux-gnu-gprof
riscv64-unknown-linux-gnu-ld
riscv64-unknown-linux-gnu-ld.bfd
riscv64-unknown-linux-gnu-lto-dump
riscv64-unknown-linux-gnu-nm
riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-objdump
riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-run
riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-strip
```
I have included the above folder to $PATH. Neverthelss, I am still getting the same error.
Did I do something wrong with my installation instructions of the riscv-gnu-toolchain?
How should I modify them in order to build the `riscv64-unknown-elf toolchain`?

Thank you in advance for your response and your help.

Kind regards,
Nassos

0ena, 2020-07-31:
My eye overpassed the last sentence in the https://github.com/riscv/riscv-gnu-toolchain#installation-newlib. I should have installed Newlib for the `riscv64-unknown-elf toolchain` and not the Linux one. I am reconfiguring and remaking the toolchain inside the same installation folder using the following commands:

```
make clean 
../configure --prefix=INSTALLATION_PATH_NAME --enable-multilib 
make newlib
```
The installation seems to continue without any problem so far (it takes some time to be completed).
If it fails, I will completely delete the installation folder and make the installation from scratch.

muhammadsalmanafzal, 2020-12-14:
What @jim-wilson said is **maybe** true, you need to compile all the components separately. For the error which is due to riscv-pk, I was able to remove it and compile it perfectly by following the follow steps:

1. Compile **64-bit** toolchain from [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain) repo.
2. Add gcc to you **PATH** by adding following lines to your **~/.\<shell\>rc**
```
export example1=/path/to/bin/of/toolchain
export example2=/path/to/toolchain
export PATH="$example1:$PATH"
export PATH="$example2:$PATH"
```
3. Restart bash.
4. Now try to build it again by following instructions in the standard [riscv-pk](https://github.com/riscv/riscv-pk) repo.

Hope this helps. If you have any more issues, do reach out. Thanks.

ShanShuiXiLiu, 2021-07-25:
i has had "riscv64-unknown-elf-gcc" in "/opt/riscv/bin" and also export RISCV=/opt/riscv.
but i meet the error "/pk/pk.c:139:3: error: both arguments to '__builtin___clear_cache' must be pointers
  139 |   __clear_cache(0,0);".
please tell me the way to resolve the problem if you met or do it

jim-wilson, 2021-07-25:
riscv-tools is not maintained and is not expected to build.  I don't recommend using it.

If you really want to try, then updating submodules may make it buildable.  cd into riscv-pk and run git log and you will see that you have a March 2019 version of riscv-pk which is why it doesn't build.  This problem was fixed Apr 28 2021 in riscv-pk, but since you have a 2+ year old version of it you don't have the fix.  If you cd into riscv-pk, then do "git checkout master" and then "git pull" you can get an up to date copy of riscv-pk.  Likewise in the other subdirs.  Just don't update submodules at the toplevel or you will get the old broken versions of subdirs again.

ZhaoZiHao1998, 2022-05-11:
> @jim-wilson说的**可能是**真的，你需要单独编译所有组件。对于由于riscv-pk引起的错误，我能够按照以下步骤将其删除并完美地编译它：
> 
> 1. 从 [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain) 存储库编译 **64 位**工具链。
> 2. 通过将以下行添加到 **~/.<shell>rc 中**，将 gcc 添加到 **PATH** 中
> 
> ```
> export example1=/path/to/bin/of/toolchain
> export example2=/path/to/toolchain
> export PATH="$example1:$PATH"
> export PATH="$example2:$PATH"
> ```
> 
> 3. 重新启动 bash。
> 4. 现在尝试按照标准[riscv-pk](https://github.com/riscv/riscv-pk)存储库中的说明再次构建它。
> 
> 希望这有帮助。如果您还有其他问题，请与我们联系。谢谢。

Thanks，my problem was solved by using your method