riscv-software-src/riscv-tools #293: Assembler messages: Fatal error: invalid -march= option: 'rv32imc'?

lianggongkai, 2019-09-13:
Hello, I encountered this problem. I have installed complile tools as the github's instructions.
But when I complie the project, I encounter this message , "Fatal error: invalid -march= option: 'rv32imc'"。Now I have no idea how to go on. Thank you for your help.

jim-wilson, 2019-09-13:
You haven't provided enough info for us to understand what the problem is.  We usually need enough info to reproduce in order to understand the problem.  You didn't give the gcc command.   You didn't give us enough info to understand if this is a gcc error or an assembler error etc.

Also, riscv-tools isn't the compiler.  Where did you get the compiler?

Try adding -v to the gcc command to get verbose output from gcc, which might point to the problem.

lianggongkai, 2019-09-16:
OK, the compile cmd is "riscv64-unknown-elf-gcc main.c -march=rv32imc -mabi=ilp32", then the output message is "Assembler messages: Fatal error: invalid -march= option: 'rv32imc' ". Just a single line of command to compile a hello world.

jim-wilson, 2019-09-16:
The "Assembler messages" is important.  It suggests that you don't have a riscv assembler.  Rerun gcc with -v to see what it is doing.  This will show you which assembler is being used.

Also try looking at your install tree.  I would expect that the compiler is $prefix/bin/riscv64-unknown-elf-gcc and the assembler is $prefix/riscv64-unknown-elf/bin/as.

lianggongkai, 2019-09-17:
Message is as below. That seems OK.
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/kunlun2/bin/../libexec/gcc/riscv64-unknown-elf/7.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/syuan/project/sifive/freedom-e-sdk/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/syuan/project/sifive/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/syuan/project/sifive/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=/home/syuan/project/sifive/freedom-e-sdk/riscv-gnu-toolchain/riscv-gcc --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medany'
Thread model: single
gcc version 7.2.0 (GCC)

kito-cheng, 2019-09-17:
Could you compile a program with `-v`? That will show which assembler is invoked.

lianggongkai, 2019-09-17:
I can't compile a program properly. Just like the message blow.

riscv64-unknown-elf-gcc -v main.c 
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/kunlun2/bin/../libexec/gcc/riscv64-unknown-elf/7.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/syuan/project/sifive/freedom-e-sdk/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/syuan/project/sifive/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/syuan/project/sifive/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=/home/syuan/project/sifive/freedom-e-sdk/riscv-gnu-toolchain/riscv-gcc --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medany'
Thread model: single
gcc version 7.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-march=rv64imafdc' '-mabi=lp64d'
 /opt/kunlun2/bin/../libexec/gcc/riscv64-unknown-elf/7.2.0/cc1 -quiet -v -imultilib rv64imafdc/lp64d -iprefix /opt/kunlun2/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/ -isysroot /opt/kunlun2/bin/../riscv64-unknown-elf main.c -quiet -dumpbase main.c -march=rv64imafdc -mabi=lp64d -auxbase main -version -o /tmp/ccVsTuIc.s
GNU C11 (GCC) version 7.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 5.4.0 20160609, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version none
warning: GMP header version 6.1.0 differs from library version 6.1.2.
warning: MPC header version 1.0.3 differs from library version 1.1.0.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring duplicate directory "/opt/kunlun2/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/7.2.0/include"
ignoring nonexistent directory "/opt/kunlun2/bin/../riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/kunlun2/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/7.2.0/include-fixed"
ignoring duplicate directory "/opt/kunlun2/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/include"
ignoring duplicate directory "/opt/kunlun2/bin/../riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/kunlun2/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/include
 /opt/kunlun2/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/include-fixed
 /opt/kunlun2/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C11 (GCC) version 7.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 5.4.0 20160609, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version none
warning: GMP header version 6.1.0 differs from library version 6.1.2.
warning: MPC header version 1.0.3 differs from library version 1.1.0.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: e2dd51f53b956aaa724366175a52a7a8
COLLECT_GCC_OPTIONS='-v' '-march=rv64imafdc' '-mabi=lp64d'
 as -v --traditional-format -march=rv64imafdc -mabi=lp64d -o /tmp/ccNEvJpe.o /tmp/ccVsTuIc.s
GNU assembler version 2.30 (x86_64-linux-gnu) using BFD version (GNU Binutils for Ubuntu) 2.30
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdc'

kito-cheng, 2019-09-17:
Seems like something wrong during you building toolchain, GCC invoked x86_64's assembler instead of RISC-V's one.

Could you check is there `riscv64-unknown-elf-as` in `/opt/kunlun2/bin/` and `as` in `/opt/kunlun2/riscv64-unknown-elf/bin` ?
```
as -v --traditional-format -march=rv64imafdc -mabi=lp64d -o /tmp/ccNEvJpe.o /tmp/ccVsTuIc.s
GNU assembler version 2.30 (x86_64-linux-gnu) using BFD version (GNU Binutils for Ubuntu) 2.30
```