riscv-software-src/riscv-tools #294: How could i install riscv32-unknown-elf binaries?

madhava-sahu, 2019-11-02:
I have installed riscv-gnu-toolchain in my Ubuntu 18.04.3 and got riscv64-unknown-elf binaries. But i need to install riscv32-unknown-elf binaries. please suggest how can i do that? Thanks in advance.

jim-wilson, 2019-11-02:
If you use --enable-multilib, then you can use a rv64 compiler as a rv32 compiler by adding -march=X -mabi=Y options.  This is the way the toolchain is intended to be used.

But if you really need a rv32 compiler you can specify --with-arch=rv32gc or whatever architecture you want when configuring.  You will also need to specify --with-abi=ilp32d or whatever abi you want.

madhava-sahu, 2019-11-05:
Hi,
I tried with above mentioned inputs..

./configure --prefix=/home/system/Desktop/elf32-toolchain --with-arch=rv32gc --with-abi=ilp32d
make

As a result, got the 32bit binaries. After executing a sample program, a linker error message coming.
please find the attachment below.
[image]

kito-cheng, 2019-11-05:
Seems like you invoke `riscv32-unknown-elf-gcc` from `pulp-toolchain` instead of `elf32-toolchiain`? Try `./riscv32-unknown-elf-gcc -o sample sample.c ` at `~/Desktop/elf32-toolchain/bin` again?

madhava-sahu, 2019-11-23:
@kito-cheng thank you very much. seems like it is the problem.after that done, i am getting ../pk/elf.c:40: assertion failed: IS_ELF64(eh)
error as pk is built for 64bit.

jim-wilson, 2019-12-02:
pk/elf.c does

#if __riscv_xlen == 64
  assert(IS_ELF64(eh));
#else
  assert(IS_ELF32(eh));
#endif

so you should get a 32-bit pk if you use a riscv32 compiler to build it.

The toplevel build.sh has
CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-elf
so using --host=riscv32-unknown-elf should work.  There is no default support for building a 32-bit pk so you will have to hack the build to get it.