riscv-software-src/riscv-tools #296: Repo status and deprecated note

towoe, 2019-11-13:
There are some comments in issues https://github.com/riscv/riscv-tools/issues/290#issuecomment-536127889 and https://github.com/riscv/riscv-tools/issues/278#issuecomment-469447363 which say _riscv-tools_ should not be used.
In the case that this is correct, it might be good to add a note to the README stating that status.
I can create a PR for that, but do not want to presume the repo condition.

jim-wilson, 2019-11-13:
Those comments are all referring to the fact that riscv-tools used to contain a broken compiler that was generating bogus compiler bug reports, which was a major problem for me, the compiler maintainer.  But the compiler was removed, so that problem has been fixed.

There is still the problem that no one is actively maintaining riscv-tools.  It used to be that rocket-chip included riscv-tools, and the only time anyone ever updated riscv-tools was when rocket-chip was broken and needed a fix merged into riscv-tools.  But now I think that rocket-chip no longer uses riscv-tools, and hence no one is updating it anymore.  I see no updates since April, when the compiler was removed.

As for the submodules, riscv-pk has been deprecated and replaced with opensbi, though it may still useful for some applications.  I think the others are still being maintained, it is just that no one is pulling those updates into riscv-tools, making riscv-tools pretty useless.  Vector extension support was added to riscv-isa-sim for instance, but it isn't in riscv-tools.

jim-wilson, 2019-11-13:
Actually, for riscv-pk I was thinking bbl, which is deprecated, but there is also the pk proxy kernel that may still be in use.

towoe, 2019-11-14:
Thanks @jim-wilson for the explanation.

As no one is updating the repo I think adding a warning to start of the README feels like a sensible addition. Maybe even to the repo's description (which I can not do). Or maybe it makes sense to archive the repo? As all the submodules live on their own, this repo might not provide any benefit.

I know that looking at the date of the last commit gives already an indication, but for me that was happening on the second glance, after I cloned the repo and wanted to get started.