riscv-software-src/riscv-tools #301: How many bits are used as virtual addresses on riscv-tools？

TATAyu0413, 2019-12-17:
''For RV64, two paged virtual-memory schemes are defined: Sv39 and Sv48.''
and how many bits are used as virtual addresses on riscv-tools？thank you so much!

aswaterman, 2019-12-17:
The user-space code is agnostic to that parameter; pointers are 64 bits in the current RV64 ABIs.  ABI-compliant binaries work in both Sv39 and Sv48, and they can use all of the address space that the OS has chosen to avail.

TATAyu0413, 2019-12-17:
> The user-space code is agnostic to that parameter; pointers are 64 bits in the current RV64 ABIs. ABI-compliant binaries work in both Sv39 and Sv48, and they can use all of the address space that the OS has chosen to avail.

I'm confused about virtual address in Spike RISC-V ISA Simulator.
In the file /riscv-tools/riscv-isa-sim/riscv/dts.cc, "mmu-type = \"riscv," << (procs[i]->get_max_xlen() <= 32 ? "sv32" : "sv48") << "\";\n", does it mean Sv48 is used for RV64 in Spike?

closed 2019-12-25 by TATAyu0413