riscv-software-src/riscv-tools #302: Variable tracking size exceeded error on running build.sh while compiling spike_main/disasm.cc

anirudhajith, 2020-01-20:
I run Ubuntu 18.04. On a fresh clone of riscv-tools, I followed (as far as I can tell) the instructions in `README.md`. On running `./build.sh` as mentioned, I got the following output:
```
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
ar: `u' modifier ignored since `D' is the default (see `U')
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
Installing project riscv-openocd
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'

Removing existing riscv-isa-sim/build directory
Configuring project riscv-isa-sim
Building project riscv-isa-sim
../spike_main/disasm.cc: In constructor ‘disassembler_t::disassembler_t(int)’:
../spike_main/disasm.cc:275:1: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 disassembler_t::disassembler_t(int xlen)
 ^
Installing project riscv-isa-sim

Removing existing riscv-pk/build directory
Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [file.o] Error 1
```
As you can see, `riscv-openocd` compiled with warnings but `riscv-isa-sim` and `riscv-pk` seem to have bombed pretty conclusively.

jim-wilson, 2020-01-20:
The variable tracking size limit warning can be ignored.  It just means that we had to emit less accurate debug info because the code was too complicated for better debug info variable tracking.

The mcmodel error is one you get when you accidentally use an x86_64 compiler to try to compile risc-v code.  Risc-v gcc accepts medany but x86_64 gcc does not.  You probably forgot to set an environment variable, or set it wrong.  Or maybe don't have a riscv compiler available.  Check the RISCV environment variable.  If $RISCV/bin/riscv64-unknown-elf-gcc doesn't exist, then you won't be able to build riscv-tools.

muhammadsalmanafzal, 2020-12-14:
What @jim-wilson said is **maybe** true, you need to compile all the components separately. For the error which is due to riscv-pk, I was able to remove it and compile it perfectly by following the follow steps:

1. Compile **64-bit** toolchain from [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain) repo.
2. Add gcc to you **PATH** by adding following lines to your **~/.\<shell\>rc**
```
export example1=/path/to/bin/of/toolchain
export example2=/path/to/toolchain
export PATH="$example1:$PATH"
export PATH="$example2:$PATH"
```
3. Restart bash.
4. Now try to build it again by following instructions in the standard [riscv-pk](https://github.com/riscv/riscv-pk) repo.

Hope this helps. If you have any more issues, do reach out. Thanks.