riscv-software-src/riscv-tools #303: How could i compile spike and pk in 32bit?

qs132, 2020-03-08:
I want to emulate an executable under a 32-bit architecture.So I changed the content of the riscv-tools/build.sh file to：

build_project riscv-isa-sim --prefix=$RISCV
CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv32-unknown-elf
build_project riscv-tests --prefix=$RISCV/riscv32-unknown-elf

then make it.
I checked the Makefile in the riscv-tools/riscv-pk/build directory,it had been changed.

CC            := riscv32-unknown-elf-gcc
READELF       := riscv32-unknown-elf-readelf
OBJCOPY       := riscv32-unknown-elf-objcopy
CFLAGS        := -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE $(CFLAGS) $(march) $(mabi) -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -fno-stack-protector -U_FORTIFY_SOURCE
BBL_PAYLOAD   := dummy_payload
COMPILE       := $(CC) -MMD -MP $(CFLAGS) \
                 $(sprojs_include)

Does this mean that pk has been compiled on a 32-bit architecture?
But when i test it,it does not work!

heyman@heyman-virtual-machine:~/gcc_test$ riscv32-unknown-elf-gcc -o hello32 hello.c
heyman@heyman-virtual-machine:~/gcc_test$ spike pk hello32
bbl loader
../pk/elf.c:40: assertion failed: IS_ELF64(eh)
heyman@heyman-virtual-machine:~/gcc_test$ riscv64-unknown-elf-gcc -o hello64 hello.c
heyman@heyman-virtual-machine:~/gcc_test$ spike pk hello64
bbl loader
hello woldheyman@heyman-virtual-machine:~/gcc_test$ 

why?
How can I emulate in a 32-bit architecture ?

aswaterman, 2020-03-08:
Try giving Spike the full path to your 32-bit pk.  It might be finding the 64-bit pk instead of the 32-bit one.  You'll also need to tell Spike what ISA to use.  Something like

    spike --isa=rv32gc $RISCV/riscv32-unknown-elf/pk hello32

qs132, 2020-03-09:
> Try giving Spike the full path to your 32-bit pk. It might be finding the 64-bit pk instead of the 32-bit one. You'll also need to tell Spike what ISA to use. Something like
> 
> ```
> spike --isa=rv32gc $RISCV/riscv32-unknown-elf/pk hello32
> ```

Got it!
Thank you very much

aswaterman, 2020-03-09:
You're welcome!

closed 2020-03-09 by aswaterman