riscv-software-src/riscv-tools #304: Issue with

Kbz6, 2020-03-24:
Hello,

I want to use the riscv-tools in ordre to work on ariane riscv project.

I followed the readme instructions for riscv-tools.

I have some issues when i start with riscv-tools, doing ./build.sh
-> I get :  

karan@karan-VirtualBox:~/FPGA/riscv-tools$ ./build.sh
Starting RISC-V Toolchain build process
./build.sh: ligne 11: automake : commande introuvable
./build.sh: ligne 11: autoconf : commande introuvable

./build.common: ligne 32: autoreconf : commande introuvable

It says that I have not the automake, autoconf and autoreconf commandes.

So, like it is indicated, I try to install the packages needed, doing :

$ sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev libusb-1.0-0-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev device-tree-compiler pkg-config libexpat-de

and I get : 
Err:1 http://fr.archive.ubuntu.com/ubuntu xenial-updates/main amd64 zlib1g amd64 1:1.2.8.dfsg-2ubuntu4.1
  404  Not Found [IP : 194.158.119.186 80]
Err:2 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libexpat1-dev amd64 2.1.0-7ubuntu0.16.04.3
  404  Not Found [IP : 194.158.119.186 80]
Err:3 http://fr.archive.ubuntu.com/ubuntu xenial-updates/main amd64 zlib1g-dev amd64 1:1.2.8.dfsg-2ubuntu4.1
  404  Not Found [IP : 194.158.119.186 80]
Err:2 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libexpat1-dev amd64 2.1.0-7ubuntu0.16.04.3
  404  Not Found [IP : 194.158.119.186 80]
E: Impossible de récupérer http://fr.archive.ubuntu.com/ubuntu/pool/main/z/zlib/zlib1g_1.2.8.dfsg-2ubuntu4.1_amd64.deb  404  Not Found [IP : 194.158.119.186 80]

E: Impossible de récupérer http://security.ubuntu.com/ubuntu/pool/main/e/expat/libexpat1-dev_2.1.0-7ubuntu0.16.04.3_amd64.deb  404  Not Found [IP : 194.158.119.186 80]

E: Impossible de récupérer http://fr.archive.ubuntu.com/ubuntu/pool/main/z/zlib/zlib1g-dev_1.2.8.dfsg-2ubuntu4.1_amd64.deb  404  Not Found [IP : 194.158.119.186 80]

E: Impossible de récupérer certaines archives, peut-être devrez-vous lancer apt-get update ou essayer avec --fix-missing ?

It says that some IP address, to get some archives, are not found.

Do you know how to fix it ?

Thank you for your help

jim-wilson, 2020-03-24:
This is a failure of apt update to connect to the fr.archive.ubuntu.com server.  This could be a temporary problem.  Try it again after waiting a few hours.  You can try visiting the site in a browser to see if you can reach it.  I am able to connect to the web site so it appears to be working.  If you can't reach the site, then there might be a networking issue, or there might be something blocking access to the site like a corporate firewall.

candasunal, 2020-08-05:
Hello All,

I have a similar problem with the ./build.sh command. Here is my input:

candasunal@LAPTOP-4ID4HG1I:~/riscv/riscv-tools$ ./build.sh

And here is my output:

Starting RISC-V Toolchain build process
./build.sh: line 11: automake: command not found
./build.sh: line 11: autoconf: command not found

build.common: line 32: autoreconf: command not found

And just like Kbz6 I tried the following command:

sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev libusb-1.0-0-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev device-tree-compiler pkg-config libexpat-de

and got the following output:

Reading package lists... Done
Building dependency tree
Reading state information... Done
Package texinfo is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  texinfo-doc-nonfree install-info info

E: Package 'texinfo' has no installation candidate
E: Unable to locate package gperf
E: Unable to locate package libexpat-de

Any idea how to fix it?

candasunal, 2020-08-07:
Hello Jim,

Thank you for your answer. I searched gperf installation on the internet and I found out that I needed to update Ubuntu in my Windows. After the update I was able to install all those packages.