riscv-software-src/riscv-tools #306: configure: error: Spike requires a little-endian host

yanjy1215, 2020-05-23:
My host is a little-endian host. however, this error shows

jim-wilson, 2020-05-24:
Check the file riscv-isa-sim/build/config.log.  You will see a message something like
configure:3825: checking whether byte ordering is bigendian
and then a lot of tests compiling small files checking to see if we get a big-endian result.  One of these tests must have failed.  This should show in the log, though you may have to cross reference with the riscv-isa-sim/configure file to see what is going on.

It isn't a question of whether the build system is little-endian, but whether the C and C++ compilers are emitting little-endian code by default.  If you have a bi-endian system, or a non-host cross compiler as $CC, then these tests could fail.

yanjy1215, 2020-05-28:
thanks! It is caused by g++ version

closed 2020-05-28 by yanjy1215