riscv-software-src/riscv-tools #312: Compile error

20171166, 2020-07-13:
Hi.

I'm currently using CentOS 7.6 and I installed the compiler from this source https://github.com/pulp-platform/pulp-riscv-gnu-toolchain
and I install this compiler at the /home/laurent01/pulp-gnu-toolchain-build so I add the PATH as follows

export PATH=$PATH:pulp-gnu-toolchain-build/bin
export RISCV="pulp-gnu-toolchain-build"

I think until here, I success.

But when I install riscv-tools from this repository,
I need to declare again RISCV variable as desired installation directory so that I just basically overwrite the variable as follows.

export RISCV=/home/laurent01/riscv/pulp-gnu-toolchian-build

But in this case as this link points out https://github.com/riscv/riscv-tools/issues/290
following error occured. gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small

So again I just do 
export RISCV="pulp-gnu-toolchain-build"

then following error message is occured.

expected an absolute directory name for --prefix : pulp-gnu-toolchain-build

So I just do
export RISCV=/home/laurent01/riscv/pulp-gnu-toolchian-build

and at the build.sh 
I replace all the $RISCV as /home/laurent01/riscv-tools-build

and I replace this line
build_project riscv-tests --prefix=$RISCV/riscv64-unknown-elf 
into as follows
build_project riscv-tests --prefix=/home/laurent01/riscv-tools-build --host=riscv64-unkown-elf

But again, following error occured. gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small

I wonder is https://github.com/pulp-platform/pulp-riscv-gnu-toolchain not work for the time installing this riscv-tools?

Thank you for reading.

jim-wilson, 2020-07-13:
I don't know if pulp-riscv-gnu-toolchain can work, but I can tell you what riscv-tools is looking for.  It wants $RISCV/bin/riscv64-unknown-elf-gcc to be a working compiler.  So RISCV needs to point at the toolchain install tree.  The use of pulp-gnu-toolchain-build looks curious above, as that looks more like a build tree than an install tree.

20171166, 2020-07-13:
> I don't know if pulp-riscv-gnu-toolchain can work, but I can tell you what riscv-tools is looking for. It wants $RISCV/bin/riscv64-unknown-elf-gcc to be a working compiler. So RISCV needs to point [...]

I don't have riscv64-unknown-elf-gcc but I have riscv32-unknown-elf-gcc under the pulp_gcc_installation_directory/bin
Is that might be a cause of the problem?

And can you specify what it means "that looks more like a build tree than an install tree" Since I'm very new to this field, I cannot understand that meaning in the exact way

jim-wilson, 2020-07-13:
Normally, you have a source tree, a build tree where you type make (which might be the source tree), and an install tree where the build artifacts are placed for end users.  In gnu tools, the --prefix= option will point at the install tree.

Yes, a riscv32-unknown-elf-gcc will not work because riscv-tools is riscv64 by default.  If you want a riscv32 riscv-tools, that is possible, but requires manually fixing some stuff to change it from riscv64 to riscv32.  Exactly how to do this varies from one dir to the next.  For riscv-pk changing the --host option in build.common should be enough I think.  There is a configure option --with-arch=rv32i (or whatever) that you can use if you want a non-default arch, and likewise a --with-abi for a non-default abi.  For riscv-tests I think it is harder as riscv64-unknown-elf-gcc is hardcoded into some scripts and Makefiles.  In riscv-isa-sim there are some subdirs that also have riscv64-unknown-elf-gcc hardcoded in them, but I don't know if these are built by default.

riscv-tools is not well maintained, and doesn't have an easy to use build system, so unfortunately there is no easy solution for this.

20171166, 2020-07-13:
> Normally, you have a source tree, a build tree where you type make (which might be the source tree), and an install tree where the build artifacts are placed for end users. In gnu tools, the [...]

I'm really thankful for your kind reply.

Actually, the very thing that I need is adding custom instruction to pulp-gcc which means that simulator is not needed.
I think it is enough to add opcodes to the structure of riscv opcodes which is located at pulp-riscv-toolchain/riscv-binutils-gdb/opcodes/riscv-opc.c But I don't know how to generate MATCH_XXX and MASK_XXX macro without simulator. Do you have any idea of it?

Since you are very skillful at this field, I think you can give a solution or at least where to start

jim-wilson, 2020-07-13:
There was a discussion about this in sw-dev recently.
https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ

20171166, 2020-07-14:
Thanks Jim. Your link is best point to start

closed 2020-08-03 by 20171166