riscv-software-src/riscv-tools #313: multiple definition

TATAyu0413, 2020-07-18:
when I do build, the issue occurs as following;
```
/home/wangyu/toolchian-gcc/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccMejApe.o: in function `tohost':
(.tohost+0x0): multiple definition of `tohost'; /tmp/cciw6Rse.o:(.sbss+0x10): first defined here
/home/wangyu/toolchian-gcc/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccMejApe.o: in function `fromhost':
(.tohost+0x40): multiple definition of `fromhost'; /tmp/cciw6Rse.o:(.sbss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [rv32ui-v-simple] Error 1
make: *** [isa] Error 2
```
Any help would be appreciated. Thank you in advance!

jim-wilson, 2020-07-18:
One change from gcc 9 to gcc 10 is that gcc 10 now makes -fno-common the default for C, same as it has been for years with C++.  This means that C files missing an extern will compile with gcc-9 but not gcc-10.  You didn't give the gcc command, so I don't know what files are being compiled here, but if you look at them, you will find one of them is missing an extern for the tohost and fromhost variable declarations.

TATAyu0413, 2020-07-19:
Thank you so much! I have solved my problem.

closed 2020-07-19 by TATAyu0413

zhuzhizhan, 2020-09-09:
> Thank you so much! I have solved my problem.

How have you solved this problem?