riscv-software-src/riscv-tools #316: Error while building riscv-tests - Multiple definitions of `tohost` and `fromhost` - riscv64-unknown-elf-gcc 10.1.0

0ena, 2020-07-31:
Hi,

I am using **riscv64-unknown-elf-gcc (GCC) 10.1.0** version and I am trying to build the **riscv-tools**.
While trying to build the **riscv-tests**, I get the following error:
```
/eda/tools/riscv/gnu-toolchain/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccjlTEE7.o: in function `tohost':
(.tohost+0x0): multiple definition of `tohost'; /tmp/cc1LzG1S.o:(.sbss+0x10): first defined here
/eda/tools/riscv/gnu-toolchain/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccjlTEE7.o: in function `fromhost':
(.tohost+0x40): multiple definition of `fromhost'; /tmp/cc1LzG1S.o:(.sbss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [rv32ui-v-simple] Error 1
make: *** [isa] Error 2
```

From a search I did, it is reported twice in the parent repo of the riscv-tests as an issue

https://github.com/riscv/riscv-tests/issues/282
https://github.com/riscv/riscv-tests/issues/286

It seems that the version of gcc is causing all the fuzz.
Are there any workarounds to finish building (besides using an older version of gcc or a precompiled toolchain)?

Thank you in advance for your responses and help.

Kind regards,
Nassos

sorear, 2020-07-31:
sounds like https://wiki.gentoo.org/wiki/Gcc_10_porting_notes/fno_common , -fcommon might work as a workaround?

jim-wilson, 2020-08-02:
It has already been fixed in riscv-tests.
https://github.com/riscv/riscv-tests/issues/286

Unfortunately, riscv-tools is rather poorly maintained, so doesn't have the fix yet.  Note that riscv-tools has a version of riscv-tests from Sept 2018, but the last change to riscv-tests was July 14, 2020.  There is almost two years worth of changes missing from riscv-tools.  The same is true for many of the other submodules.  I would suggest avoiding riscv-tools as it is mostly useless, or else if you do use riscv-tools then manually update the submodules to get up to date software.

t-cheng-ou, 2020-08-02:
I also met this issue and just resolved it.

What I did is to modify ./riscv-tests/env/v/vm.c by adding the "extern" when calling tohost and fromhost.
i.e. 
extern volatile uint64_t tohost
extern volatile uint64_t fromhost

Applying "-fcommon" does not work in my case.

0ena, 2020-08-03:
Hi,

I confirm that @t-cheng-ou 's workaround worked like a charm and the installation was completed. Thank you very much!
Nevertheless, I understand the valid points of @jim-wilson and I would like to try also his solution.  

Currently the installation path of my risc-v tools is:

RISCV=/eda/tools/riscv/riscv-tools

and the created folders inside are:

./bin
./include
./lib
./riscv64-unknown-elf
./share

If I understand your suggestion correctly, I should follow the installation instructions here https://github.com/riscv/riscv-tests and use as a prefix path in the .configure step, the risc-v tools installation path shown above:

./configure --prefix=$RISCV/

Am I correct on that? Will this update the riscv-tests submodule incorporating the latest changes?

I want to start experimenting with Ariane (https://github.com/openhwgroup/cva6) and according to their installation instruction, RISC-V tools are needed.

Thank you in advance for your time on my matter.

Kind regards,
Nassos

jim-wilson, 2020-08-03:
In theory, checking out riscv-tests and building it the same way as the riscv-tools build.sh script does should work.  However, I don't know if there are interdependencies between riscv-tests and riscv-isa-sim (aka spike) and other packages.  A new version of riscv-tests may not work with the old version of riscv-isa-sim and other packages that you have for instance.  You might need to check out and build new versions of the other packages too.

Another way to accomplish this is to use riscv-tools, but manually update the submodules.  E.g. in the riscv-tools source tree, you can do "cd riscv-tests; git pull" and you will get the current version of riscv-tests.  You then need to be careful to avoid doing a git submodule update at the top level, as that will then revert riscv-tests to the old version.  As above, this might not work unless you manually update all of the submodules.  You can see a list of them in the .gitmodules file.  Manually updating submodules might cause other problems if their build systems have changed and are no longer compatible with the riscv-tools build.sh script but that seems unlikely.

moy, 2020-08-17:
> I also met this issue and just resolved it.
> 
> What I did is to modify ./riscv-tests/env/v/vm.c by adding the "extern" when calling tohost and fromhost.
> i.e.
> extern volatile uint64_t tohost
> extern volatile uint64_t fromhost

I don't have a `./riscv-tests/env/v/vm.c` file, and according to https://github.com/riscv/riscv-tests/commit/f4e14a7a3a45ff4ef2af09696e0a2238f1227ca2 it was deleted in 2013. What am I missing?

Thanks,

moy, 2020-08-17:
> > I also met this issue and just resolved it.
> > What I did is to modify ./riscv-tests/env/v/vm.c by adding the "extern" when calling tohost and fromhost.
> > i.e.
> > extern volatile uint64_t tohost
> > extern volatile uint64_t fromhost
> 
> I don't have a `./riscv-tests/env/v/vm.c` file, and according to [riscv/riscv-tests@f4e14a7](https://github.com/riscv/riscv-tests/commit/f4e14a7a3a45ff4ef2af09696e0a2238f1227ca2) it was deleted in 2013. What am I missing?

Oops, my bad, vm.c is still there, but in a sub-sub-module (which I had deleted by mistake).

darren0h, 2020-08-31:
> Hi,
>
> I confirm that @t-cheng-ou 's workaround worked like a charm and the installation was completed. Thank you very much!
> Nevertheless, I understand the valid points of @jim-wilson and I would like [...]

Switch the riscv-tests submodule in riscv-gnu-toolchain to newer tag or master branch will let the riscv-tools build pass.

zhuzhzh, 2020-11-22:
@darren0h I am using master branch of riscv-gnu-toolchain (gcc 10.2.0). but it still reports this error. Is there anything I am missing?

jim-wilson, 2020-11-22:
There is no problem with riscv-gnu-toolchain.  it is riscv-tools that is broken; its maintenance is poor to non-existent.  You may want to stop using it.  Or you can manually update the riscv-tests submodule as riscv-tools has a 2 year old version of it.  This bug was fixed in riscv-tests months ago.
  cd riscv-tests
  git log -1
  git fetch
  git checkout master
  git pull
  git log -1
You might need to do the same in other submodules.  Updating some submodules but not others might give you inconsistent versions of submodules.