riscv-software-src/riscv-tools #317: riscv-openocd warning & error on ./build.sh

candasunal, 2020-08-08:
Hello all,

I encounter a problem with the riscv tools installation. Before installation I updated my Ubuntu on my Windows by `sudo apt-get update -y` I want to go step by step to see if I miss something just in case.

1. First I make my own directory under my username: /home/candasunal/ : `mkdir riscv`
2. Then : `cd riscv/`
3. Then I clone the git: `git clone https://github.com/riscv/riscv-tools.git` . Here is the result:
```
Cloning into 'riscv-tools'...
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 1225 (delta 0), reused 1 (delta 0), pack-reused 1222
Receiving objects: 100% (1225/1225), 736.17 KiB | 326.00 KiB/s, done.
Resolving deltas: 100% (613/613), done.
```
4. I change my directory: `cd riscv-tools`
5. Then I copy the code in Quickstart at https://github.com/riscv/riscv-tools  : `git submodule update --init --recursive` . Here is the result:
```
Submodule 'riscv-isa-sim' (https://github.com/riscv/riscv-isa-sim.git) registered for path 'riscv-isa-sim'
Submodule 'riscv-opcodes' (https://github.com/riscv/riscv-opcodes.git) registered for path 'riscv-opcodes'
Submodule 'riscv-openocd' (https://github.com/riscv/riscv-openocd.git) registered for path 'riscv-openocd'
Submodule 'riscv-pk' (https://github.com/riscv/riscv-pk.git) registered for path 'riscv-pk'
Submodule 'riscv-tests' (https://github.com/riscv/riscv-tests.git) registered for path 'riscv-tests'
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-isa-sim'...
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-opcodes'...
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-openocd'...
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-pk'...
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-tests'...
Submodule path 'riscv-isa-sim': checked out '2710fe575e7e6a4e2418224f8d254d5ca31f6c0e'
Submodule path 'riscv-opcodes': checked out '7c3db437d8d3b6961f8eb2931792eaea1c469ff3'
Submodule path 'riscv-openocd': checked out '35eed36ffdd082f5abfc16d4cc93511f6e225284'
Submodule 'jimtcl' (https://github.com/msteveb/jimtcl) registered for path 'riscv-openocd/jimtcl'
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-openocd/jimtcl'...
Submodule path 'riscv-openocd/jimtcl': checked out '51f65c6d38fbf86e1f0b036ad336761fd2ab7fa0'
Submodule path 'riscv-pk': checked out '97b683ed3cbeeb059e5ad3de8f884db734447d56'
Submodule path 'riscv-tests': checked out '79064081503b53fdb44094e32ff54a3ab20a9bf2'
Submodule 'env' (https://github.com/riscv/riscv-test-env.git) registered for path 'riscv-tests/env'
Cloning into '/home/candasunal/riscv/riscv-tools/riscv-tests/env'...
Submodule path 'riscv-tests/env': checked out '68cad7baf3ed0a4553fffd14726d24519ee1296a'
```
6. And then I install the neede packages: `sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev libusb-1.0-0-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev device-tree-compiler pkg-config libexpat-dev` . Since I have done this install before, I got zero update. The result is: 
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libexpat1-dev' instead of 'libexpat-dev'
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.15.1-3ubuntu2).
autotools-dev is already the newest version (20180224.1).
bc is already the newest version (1.07.1-2).
bison is already the newest version (2:3.0.4.dfsg-1build1).
build-essential is already the newest version (12.4ubuntu1).
device-tree-compiler is already the newest version (1.4.5-3).
flex is already the newest version (2.6.4-6).
gawk is already the newest version (1:4.1.4+dfsg-1build1).
libgmp-dev is already the newest version (2:6.1.2+dfsg-2).
libmpc-dev is already the newest version (1.1.0-1).
libmpfr-dev is already the newest version (4.0.1-1).
libtool is already the newest version (2.4.6-2).
libusb-1.0-0-dev is already the newest version (2:1.0.21-2).
patchutils is already the newest version (0.3.4-2).
pkg-config is already the newest version (0.29.1-0ubuntu2).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-0ubuntu2).
gperf is already the newest version (3.1-1).
texinfo is already the newest version (6.5.0.dfsg.1-2).
curl is already the newest version (7.58.0-2ubuntu3.9).
libexpat1-dev is already the newest version (2.2.5-3ubuntu0.2).
The following package was automatically installed and is no longer required:
  libfreetype6
[... 83 lines cut ...]
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'

Configuring project riscv-isa-sim
Building project riscv-isa-sim
../spike_main/disasm.cc: In constructor ‘disassembler_t::disassembler_t(int)’:
../spike_main/disasm.cc:275:1: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 disassembler_t::disassembler_t(int xlen)
 ^~~~~~~~~~~~~~
Installing project riscv-isa-sim
mkdir /home/candasunal/riscv/toolchain/include
mkdir /home/candasunal/riscv/toolchain/include/fesvr
mkdir /home/candasunal/riscv/toolchain/lib
mkdir /home/candasunal/riscv/toolchain/lib/pkgconfig

Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [file.o] Error 1
```
Am I missing something here?

jim-wilson, 2020-08-08:
riscv-tools is poorly maintained.  It is best to avoid it if you don't absolutely need it.  And if you do need it, you might consider building the submodules individually instead of using riscv-tools, because riscv-tools has badly out of date and broken 2 year old versions of some of its submodules.

The error comes from passing riscv compiler options to an x86_64 compiler which obviously doesn't accept them.  You need to build and install an riscv64-unknown-elf-gcc before you can build riscv-tools.  The compiler is in riscv-gnu-toolchain.  Or alternatively, you can download a pre-built one.  SiFive has its toolchain, and some 3rd party toolchains, available for download from its website.  See https://sifive.com/software

muhammadsalmanafzal, 2020-12-14:
What @jim-wilson said is **maybe** true, you need to compile all the components separately. For the error which is due to riscv-pk, I was able to remove it and compile it perfectly by following the follow steps:

1. Compile **64-bit** toolchain from [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain) repo.
2. Add gcc to you **PATH** by adding following lines to your **~/.\<shell\>rc**
```
export example1=/path/to/bin/of/toolchain
export example2=/path/to/toolchain
export PATH="$example1:$PATH"
export PATH="$example2:$PATH"
```
3. Restart bash.
4. Now try to build it again by following instructions in the standard [riscv-pk](https://github.com/riscv/riscv-pk) repo.

Hope this helps. If you have any more issues, do reach out. Thanks.