riscv-software-src/riscv-tools #323: Can not build RISCV Toolchain with ./build.sh

SeDi343, 2020-10-04:
I somehow always run into the problem, that I can not build the Toolchain using ./build.sh

[image]

SeDi343, 2020-10-04:
1. Machine:

Operating System: Ubuntu 18.04.5 LTS
Kernel: Linux 4.15.0-115-generic
Architecture: x86-64
patch state: ✓ all applicable livepatch modules inserted
patch version: 71.1
client version: 9.5.5
architecture: x86_64
cpu model: Intel(R) Xeon(R) CPU E5-1620 0 @ 3.60GHz

2. Machine:
Ubuntu on Windows
NAME="Ubuntu"
VERSION="18.04.4 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.4 LTS"
VERSION_ID="18.04"

zhuzhzh, 2020-10-08:
you need install riscv-gnu-toolchain.

SeDi343, 2020-10-13:
@zhuzhzh I installed the riscv-gnu-toolchain but it still failes with the same error

zhuzhzh, 2020-10-18:
@SeDi343  execute `which riscv64-unknown-elf-gcc`
make sure the exec path is put in $PATH

SeDi343, 2020-10-19:
@zhuzhzh you mean `which riscv64-unknown-linux-gnu-gcc`?
because that got installed with the riscv-gnu-toolchain. `which riscv64-unknown-elf-gcc` is for now leading to nothing. But that is installed with the `./build.sh`, isn't it?

zhuzhzh, 2020-11-22:
@SeDi343 you need run "make" in riscv-gnu-tools instead of "make linux".
risdv-tools needs riscv64-unknown-elf-gcc, instead of riscv64-unknown-linux-gnu-gcc