riscv-software-src/riscv-tools #324: spike pk hello  returns  "bad syscall"

zhuzhzh, 2020-10-18:
I follows the guide to build riscv-gnu-toolchain first. then build riscv-tools on CentOS7.8.

Everything looks fine.

```c
#include <stdio.h>
int main() {
    printf("hello world\n");
    return 0;
}
```

riscv64-unknown-elf-gcc -o hello helloworld.c

`which pk` and `which bbl` can be found.

```shell

# harriszh @ pc-harriszh2 in /trunk/branch/edahub/riscv-gnu-toolchain on git:master x [5:37:58] C:255
$ which spike
/home/harriszh/.local/riscv/bin/spike

# harriszh @ pc-harriszh2 in /trunk/branch/edahub/riscv-gnu-toolchain on git:master x [5:38:05] 
$ which pk
/home/harriszh/.local/riscv/bin/pk

# harriszh @ pc-harriszh2 in /trunk/branch/edahub/riscv-gnu-toolchain on git:master x [5:37:51] 
$ spike -m128 pk hello 
bbl loader
terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #90463429510455458

```

I tried with qemu-riscv64. It's okay.
```shell
# harriszh @ pc-harriszh2 in /trunk/branch/edahub/riscv-gnu-toolchain on git:master x [5:38:07] 
$ qemu-riscv64 hello
hello world!
```

Do you know why spike pk doesn't work?

oaltinay, 2020-11-01:
I had the same issue. I found out that latest riscv64-unknown-elf-gcc version is not compatible with spike which is in this repo. You can use [SiFive](https://www.sifive.com/software)'s prebuilt GNU Embedded Tool chain. It's version working with spike pk without an error.

shitianbushuijiao, 2020-12-14:
I had the same issue，And solved it by https://github.com/riscv/riscv-pk/issues/170 which explains tree-loop-distribute-patterns

zhuzhzh, 2021-01-28:
@shitianbushuijiao thanks, it works with the change as riscv/riscv-pk#170

pavitrar20, 2022-11-08:
I still encounter this error. Using da01ba455ce3802ffa84fdca3a089079996dbfc3 commit of(HEAD -> rvv-next, origin/rvv-next)
I need to test some simple vector instructions so using rvv-next branch
riscv-gnu-toolchain$ git submodule status
9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc (glibc-2.33)
-85e0e3519655220688e757b9d5bfd314923548bd musl
415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)
-553032db17440f8de011390e5a1cfddd13751b0b qemu
b51c2fec1da205ea3e7354cbb3e253018d64873c riscv-binutils (binutils-2_39)
-4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu
+eb96155f49deefcde0b054f6c02aa67c607ac482 riscv-gcc (remotes/origin/riscv-gcc-rvv-next)
5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)

But am unable to execute prorams (sclara and vector) with spike pk, 

I have tried some permuations between riscv32-unknown-elf-gcc andriscv64-unknown-elf-gcc

But although program always compiles, I encounter error described here

Compiled a simple program with riscv32-unknown-elf-gcc -march=rv32gcv hello.c -o hello

Invoked PK in two ways - 
1. Specified explcit path -spike $RISC/riscv32-unknown-elf/bin/pk hello --> result hang
2. . spike pk hello -bbl loader
../pk/elf.c:40: assertion failed: IS_ELF64(eh)