riscv-software-src/riscv-tools #325: riscv64-unknown-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory

lxu28973, 2020-10-23:
Why I got : Configured with: ../configure --build=x86_64-linux-gnu
Shouldn't x86_64-linux-gnu be riscv64-unknown-elf-gcc?

lei@lei-virtual-machine:~/G/rocket-chip/emulator$ riscv64-unknown-elf-gcc -v helloworld.c -o helloworld
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 9.3.0 () 
COLLECT_GCC_OPTIONS='-v' '-o' 'helloworld' '-march=rv64imafdc' '-mabi=lp64d'
 cc1 -quiet -v -imultilib rv64imafdc/lp64d helloworld.c -quiet -dumpbase helloworld.c -march=rv64imafdc -mabi=lp64d -auxbase helloworld -version -o /tmp/ccDh8DG7.s
riscv64-unknown-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.

jim-wilson, 2020-10-23:
Build is the machine that you used to build the compiler.  Host is the machine that the compiler will run on.  Target is the machine that the compiler produces code for.  So for instance if you want to use an x86_64-linux machine to build a MS windows cross compiler for riscv64-elf, then build is x86_64-linux, host is ms windows, and target is riscv64-elf.

The prefix is /usr.  This is unusual for hand built compilers.  Did you maybe install an OS package to get this compiler?  The cc1 error suggests that maybe you are missing the package that contains the cc1 binary.

If you built this yourself, then there might have been a problem during the build that you missed.  Note that installing into /usr will make it hard to remove a broken compiler.  It is best to use a dir that you can rm -rf if something goes wrong.  This is OK for an OS package as the package manager will keep track of what files it installed so that they can be safely removed.  But manual builds don't have this.

lxu28973, 2020-10-24:
Thank you for your cordial answer. Yes, there is something wrong when I built the compiler. I rm and rebuild it. Now it works.

closed 2020-10-24 by lxu28973