riscv-software-src/riscv-tools #328: Error while build riscv-tool , riscv-isa-sim

vijayganesh, 2020-12-09:
while installing riscv-tool from build i got error 

Configuring project riscv-isa-sim
Building project riscv-isa-sim

./fesvr/dtm.cc: In member function ‘uint32_t dtm_t::get_xlen()’:
../fesvr/dtm.cc:488:16: error: ‘runtime_error’ is not a member of ‘std’
  488 |     throw std::runtime_error("FESVR DTM Does not support 128-bit");
      |                ^~~~~~~~~~~~~
../fesvr/dtm.cc:505:14: error: ‘runtime_error’ is not a member of ‘std’
  505 |   throw std::runtime_error("FESVR DTM can't determine XLEN. Aborting");
      |              ^~~~~~~~~~~~~
../fesvr/dtm.cc:506:1: warning: control reaches end of non-void function [-Wreturn-type]
  506 | }
      | ^
can anyone help me.

jim-wilson, 2020-12-09:
In the SiFive forums you mentioned Ubuntu 20.10.  It builds for me on Ubuntu 20.04.  I don't have 20.10 to test.  fesvr is built by the native compiler, this suggests a problem with your native C++ compiler.  You might be able to find some useful info in riscv-isa-sim/build/build.log.  Verify that it is being built with "g++" and "-std=c+=11".  Check that "g++" is actually an x86_64-linux compiler, assuming that your host is x86_64-linux.  Sometimes people put a riscv-elf gcc/g++ on their path which will break compilation of native x86 programs.  So try "which g++" and "g++ -v" to verify that it is a native compiler.

vijayganesh, 2020-12-10:
@jim-wilson  thanks for reply. 
my gcc location is /usr/bin/g++ 
g++ -v  gives me

COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/10/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 10.2.0-13ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-10-JvwpWM/gcc-10-10.2.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-10-JvwpWM/gcc-10-10.2.0/debian/tmp-gcn/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Ubuntu 10.2.0-13ubuntu1) 

and in build.log -std=c++11 is there.

g++ -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/chips/riscv/toolchain\" -I. -I.. -I../fesvr -I../riscv -I../dumm
y_rocc -I../softfloat -I../spike_main -fPIC -c ../fesvr/elfloader.cc
g++ -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/chips/riscv/toolchain\" -I. -I.. -I../fesvr -I../riscv -I../dumm
y_rocc -I../softfloat -I../spike_main -fPIC -c ../fesvr/htif.cc
g++ -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/chips/riscv/toolchain\" -I. -I.. -I../fesvr -I../riscv -I../dumm
y_rocc -I../softfloat -I../spike_main -fPIC -c ../fesvr/memif.cc
g++ -MMD -MP  -Wall -Wno-unused -g -O2 **-std=c++11** -DPREFIX=\"/chips/riscv/toolchain\" -I. -I.. -I../fesvr -I../riscv -I../dumm
y_rocc -I../softfloat -I../spike_main -fPIC -c ../fesvr/dtm.cc

jim-wilson, 2020-12-10:
I don't know how you are hitting this problem.  Since I can't reproduce it, I can't debug it.  You could try to debug it.  Try extracting the failing g++ command from the build.log file and running it by hand.  Does it fail?  I would expect it to.  Try adding --save-temps to the command.  This will create a .ii file which is the preprocessor output.  Check to make sure that it mentions /usr/include/c++/$gccversion/stdexcept.  And check to make sure that there is a definition of std::runtime_error in the .ii file that should be in the stdexcept section.

vijayganesh, 2020-12-10:
Hi Jim-wilson, It is solved by adding #include <stdexcept>  in file dtm.cc and riscv/devices.h file.

closed 2020-12-10 by vijayganesh