riscv-software-src/riscv-tools #329: ERROR When Compiling C File

Louisyoung056, 2020-12-30:
Hi, 
_What I Did_
I have installed **riscv-gnu-toolchain** on my CentOS7 virtual machine. 
I followed instructions on https://github.com/riscv/riscv-gnu-toolchain and successfully installed into **riscv64** folder. 
Then I installed elf2hex from github.com/sifive/elf2hex.git.
_Problem_
It worked fine when I try to run RISCV Assembly Code. But it gave error message like this when compiling C Code:
[image]

Could you please help me with this?
Thanks!

jim-wilson, 2020-12-30:
The compiler will accept any reasonable arch/abi combination, but the precompiled libraries will only be compiled for the default arch/abi specified at configure time.  You can getting an error linking against the precompiled libraries as they were built with a different arch/abi then you specified on the command line.

You can use --enable-multilib at configure time which will build multiple arch/abi versions of the precompiled libraries, but it only builds a subset of them.  There are far too many possibilities to be reasonable to build all of them.  So this will help only if you are using one of the arch/abi combinations in the default set.  rv32imc/ilp32 is not in the default set, but we have a re-use rule that tells gcc to link witih the rv32im/ilp32 precompiled library.  If that is OK for you, then --enable-multilib will work for you.

Otherwise, you should specify --with-arch= and --with-abi= at configure time to get a toolchain that directly supports your arch/abi choice directly.

Louisyoung056, 2020-12-31:
> The compiler will accept any reasonable arch/abi combination, but the precompiled libraries will only be compiled for the default arch/abi specified at configure time. You can getting an error [...]

Thanks for the advice! Problem solved using `--enable-multilib`.

closed 2020-12-31 by Louisyoung056