riscv-software-src/riscv-tools #331: riscv-isa-sim/fesvr/dtm.cc needs #include <stdexcept>

mwette, 2021-02-03:
I was getting
```
 error: 'runtime_error' is not a member of 'std'
```
I fixed by adding
```
#include <stdexcept>
```
to fesvr/dtm.cd

aswaterman, 2021-02-03:
Can you make a pull request?

aswaterman, 2021-02-03:
Looks like this has already been fixed upstream in `riscv-isa-sim`.  `riscv-tools` itself isn't really supported anymore, so your one-off fix is probably the right thing to do.

jim-wilson, 2021-02-03:
There are multiple problems with riscv-tools due to lack of maintenance.  Note that you have almost 2 year old versions of the submodules, which means you are missing almost all of the development work from the last two years.  I would suggest either not using it.  Or if you insist on using it, then manually update all of the submodules to the most recent version of them, e.g. cd into each subdir, git checkout master then git pull.  If you do another update at the top level, then that will reset the submodules, so you may need to manually set them to top of master again.  Updating submodules to the top of their master branches might not work if there are inconsistencies between the submodules that haven't been fixed yet though.

There is a useful trick Kito added to riscv-gnu-toolchain recently.  Submodules normally use a commit id.  So updating the submodule project does not cause the master project to also be updated.  But you can specify a branch name in the .gitmodules file and things work a little better.  If you use git submodule update --remote then it will check out the branch instead of the commit id.  If you use git submodule update as usual, then it only checks out the original commit id same as it works now.  But at least it makes it easier for people to get newer versions of submodules if they want them.  You could add branch info to the gitmodules file, and then add docs explaining how to get more up to date sources by using --remote.  We already did this for riscv-gnu-toolchain.  See for instance
https://github.com/riscv/riscv-gnu-toolchain/pull/816/files

mwette, 2021-02-03:
Thanks.   I would rather move to the new stuff.  Is there a list of packages I need for cross-compiling asm and C and running on a simulator?

closed 2021-02-03 by mwette