riscv-software-src/riscv-tools #332: SwiftShader for RISC-V32

amithmath, 2021-02-04:
I am trying to build: https://github.com/google/swiftshader  for RISC-V32 but SwiftShader requires CMAKE_CXX_STANDARD 17/c++17. I am wondering, how do I support this feature in RISC-V32?

Actually, I did google C++17 RISC-V32, I got following links for RISC-V64:

https://packages.debian.org/buster/g++-8-riscv64-linux-gnu

https://gnutoolchains.com/risc-v/

I am wondering whether is there gcc7 for RISC-V32?

jim-wilson, 2021-02-04:
There is a stable gcc-7.3 for RISC-V but you are better off using a newer version.  The current supported version of riscv/riscv-gnu-toolchain is gcc-10.2 for instance.  Why do you need or want gcc-7?  Usually people specify a minimum version, and any version after the specified minimum is also OK.

amithmath, 2021-02-04:
I am not so specific about gcc-7.3 as long as the gcc supports C++17 it's fine.  So if install, from https://github.com/riscv/riscv-gnu-toolchain will I get gcc-10.2 with C++17 features? Or please let me know in which script  I have to specify gcc latest version to build latest compiler....

jim-wilson, 2021-02-04:
I don't follow the C++ standard.  If the software builds with gcc, then it should build with 10.2 as that is the latest FSf GCC release.  Keep in mind that linux software may not compile with a embedded elf compoiler, if it needs library functions not supported by the embedded elf toolchain.

riscv/riscv-gnu-toolchain is source and needs to be built.  If you don't have experience building stuff it could take some time to get the build to work.  There are build instructions in the README.md file.  You could also try a precompiled package.  Most linux distros provide a riscv64-linux compiler if you have a recent enough version of the distro.  SiFive has precompiled toolchain on their web site on the software page, but only for embedded elf.

amithmath, 2021-02-04:
If you could tell me, whether this repo: https://github.com/google/swiftshader successfully compiles in latest risc-v32 gcc 10.2 or not that will be very helpful. This  is for 3D engine with OpenGL support.

Many Thanks.

jim-wilson, 2021-02-04:
There is no way for me to know if some random software package compiles for RISC-V.

amithmath, 2021-02-11:
I am looking for ../lib64/librt.so I did 'grep' and 'find' in risc-v root
directory, I couldn't get. Can you please tell me where to/how to generate
this file?

amithmath, 2021-02-11:
I am looking for RISC-V32 with gcc version 9.2.0

jim-wilson, 2021-02-11:
This was also askied in riscv-gnu-toolchain issue 840.  An embedded elf compiler can't support librt because it requires OS support, and the embedded elf toolchain assumes no OS.  The linux compiler does support it.

amithmath, 2021-02-13:
How to enable elf toolchain to use linux? Or Is  risc-v32 hardware architecture different for elf and linux?

jim-wilson, 2021-02-13:
An embedded elf toolchain assumes you are running on bare metal.  A linux toolchain assumes you are running on hardware running linux.  Both will produce programs that can run on a simulator.  A linux system requires an MMU which not all hardware has, so it won't run on the lowest cost boards.  Otherwise, the hardware is the same.

A toolchain can be either embedded elf or linux.  You decide when you configure and build the toolchain.  You can use --enable-linux to get a linux toolchain by default.  Or you can use "make linux" to build a linux toolchain.  See the README.md file.

Jamesmelodyash, 2021-02-16:
I am wondering, does Linux run on manycore, let's say 128 RISC-V32 cores?

amithmath, 2021-02-16:
Yes, that was the good question, does does Linux run on manycore?

amithmath, 2021-02-16:
I think in order to run manycore it requires glibc rather than newlib and some modifications in DRAM libraries?

jim-wilson, 2021-02-16:
New issues for new questions please.  For manycore support, you need to ask a linux kernel expert.  I'm a compiler expert.  I only know that it works for 4 cores because that is how many cores the HiFive Unleashed has.

amithmath, 2021-02-17:
It just clicked my mind, since you are compiler expert and  it works for 4 cores, same concepts can be applied to manycores? If you could point to me to HiFive Unleashed 4 cores compiler/linux kernel github repo, that will be great since HiFive has many repos...

jim-wilson, 2021-02-17:
Everything you need is upstream.  Link kernel sources are on kernel.org.  But we do have our own open embedded/yocto layer which is github.com/sifive/meta-sifive.