riscv-software-src/riscv-tools #333: problem in ./buildsh || not able to build project riscv-openocd

AnushaGoel14, 2021-02-06:
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
In file included from ../src/flash/common.h:21,
                 from ../src/flash/nor/core.h:25,
                 from ../src/flash/nor/imp.h:22,
                 from ../src/flash/nor/jtagspi.c:22:
../src/flash/nor/jtagspi.c: In function ‘jtagspi_write_enable’:
../src/helper/log.h:130:2: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  130 |  log_printf_lf(LOG_LVL_ERROR, __FILE__, __LINE__, __func__, expr)
      |  ^~~~~~~~~~~~~
../src/flash/nor/jtagspi.c:255:11: note: ‘status’ was declared here
  255 |  uint32_t status;
      |           ^~~~~~
../src/flash/nor/jtagspi.c: In function ‘jtagspi_wait’:
../src/flash/nor/jtagspi.c:242:15: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  242 |   if ((status & SPIFLASH_BSY_BIT) == 0) {
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-011.c:18:
../src/target/riscv/riscv-011.c: In function ‘dbus_scan’:
../src/helper/binarybuffer.h:93:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   93 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-013.c:18:
../src/target/riscv/riscv-013.c: In function ‘dmi_scan.constprop’:
../src/helper/binarybuffer.h:55:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   55 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2897: src/openocd] Error 1
gmake[1]: *** [Makefile:4181: all-recursive] Error 1
gmake: *** [Makefile:1953: all] Error 2

jim-wilson, 2021-02-06:
riscv-tools is not expected to build.  It hasn't been maintained in 2 years, and you may not want to use it.

This particular problem suggests that you are using gcc 10 to build riscv-tools.  gcc 10 is the first gcc version to make -fno-common the default.  That exposes bugs in old code that don't declare variables properly.  Such program bugs have probably already been fixed, but the fixes have not pulled into riscv-tools, and may never be pulled in.

You can try fixing this by adding -fcommon to CFLAGS when building openocd, but if it is a problem with a library that might not help.  Or you could try using gcc-9 instead of gcc-10 to compile it.

You can also try fixing this by using a more recent version of openocd.  If you cd into the dir and run git log you will see that you have a version from Sept 2018.  If you checkout the riscv branch and pull you will get the current version of it.  You may also need to update the other subdirs, as you have 2 or 3 year old versions of them also.  You may need to check the default branch name for each subdir as they may be different, most are probably master but at least one (openocd) isn't.  Once you have updated all subdirs, you will have to check for incompatilities between the subdirs since there is probably no one else doing that.

jim-wilson, 2021-03-09:
You can get the gcc version by using "gcc -v".

Whether you can use an older gcc versions depends on what system you are using to build on.  Some systems support multiple gcc versions.  Some don't.

Setting CFLAGS depends on the shell you are using.  Assuming bash, you can probably do something like
    CFLAGS="-fno-common" ./build.sh

Updating openocd is probably the better solution.  To update it do
  cd riscv-openocd
  git checkout riscv
  git pull
Then avoid doing a submodule update at the top level since this will clobber your updated riscv-openocd.

Perhaps I should be asking why you are building riscv-tools.  There is no toolchain in riscv-tools.  riscv-tools is a collection of misc tools for hardware developers.  If you want a toolchain, then you should be looking at riscv-gnu-toolchain instead.  You won't be able to build riscv-tools if you don't already have a toolchain.  If you don't have a lot of experience building software, and all you want is a toolchain, then you may be able to download pre-built binaries.  SiFive has toolchain binaries on their web site for instance on the software page.

RaoShri, 2021-03-10:
Hi Wilson,
I am trying to build a custom Rocketchip using Chipyard. I encountered a problem while compiling, where I was pointed out that the issue was due to me not having built the riscv-tools. 
I am using kali linux, I use zsh, gcc-10

I am still getting errors after updating riscv-openocd:

└─$ ./build.sh CFLAGS="-fno-common"                                                                                                                                                                                                      
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
../configure: line 4341: gl_EARLY: command not found
../configure: line 13831: gl_INIT: command not found
config.status: error: cannot find input file: `src/gnulib/Makefile.in'

I am guessing this is an issue with gnu compiler?? I have compiled and built riscv64-unknown-linux-gnu
Any help is valued.

Thanks a lot!
Shrinidhi

jim-wilson, 2021-03-11:
You must have a riscv64-unknown-elf toolchain to build riscv-tools.

What version of riscv-openocd do you have?  The errors given don't map to the riscv-openocd version in riscv-tools.  It looks like you might have an updated version, which means you might actually need to ask the riscv-openocd folks not the riscv-tools folks for help.

gl_EARLY and gl_INIT come from the src/m4 dir, which suggests a problem with m4, and maybe indirectly with autoconf or automake.

src/gnulib has a Makefile.am but not a Makefile.in, which is maybe a problem with a missing tool on your system (automake) or maybe a problem with a broken Makefile in riscv-openocd.

Or maybe even a git issue where you checked out a new version of rixscv-openocd but didn't update submodules inside riscv-openocd after changing the branch.  Try runnign "git status" in the riscv-openocd dir.

Note that in general riscv-tools is hard to build, and the lack of maintenance means it is no longer very useful.  You may not be able to build it unless you have some experimence building software and are willing to spend enough time to work through build issues.

RaoShri, 2021-03-12:
Hi Jim,
This is the output of git status in riscv-openocd folder:

└─$ git status                             
On branch riscv
Your branch is up to date with 'origin/riscv'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        m4/

nothing added to commit but untracked files present (use "git add" to track)