riscv-software-src/riscv-tools #336: /opt/riscv/sysroot/usr/include/sys/stat.h:25:10: fatal error: features.h: No such file or directory in building sel4 on rocket chip

swethakathir19, 2021-02-18:
I built the gcc toolchain with --enable multilib as mentioned in https://docs.sel4.systems/Hardware/rocketchip.html.
While building Sel4 test (after running ninja) I get the following error,

~/sel4/build$ ninja
[1/2] Generating ../../bbl/bbl
FAILED: bbl/bbl 
cd /home/swetha/sel4/build/apps/sel4test-driver && mkdir -p /home/swetha/sel4/build/bbl && cd /home/swetha/sel4/build/bbl && /home/swetha/sel4/projects/sel4test/../..//tools/riscv-pk/configure --quiet --host=riscv64-unknown-linux-gnu --with-arch=rv64imafdc --with-payload=/home/swetha/sel4/build/elfloader/elfloader && make -s clean && make -s > /dev/null
In file included from /home/swetha/sel4/projects/sel4test/../..//tools/riscv-pk/pk/file.h:6,
                 from /home/swetha/sel4/projects/sel4test/../..//tools/riscv-pk/pk/file.c:3:
/opt/riscv/sysroot/usr/include/sys/stat.h:25:10: fatal error: features.h: No such file or directory
   25 | #include <features.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [file.o] Error 1
ninja: build stopped: subcommand failed.

Since I am very new to this, I hope that I can get some help with this problem.

swethakathir19, 2021-02-18:
Also these are the environment variables,
swetha@swetha-Inspiron-5570:~/sel4test/build$ echo $RISCV
/opt/riscv
swetha@swetha-Inspiron-5570:~/sel4test/build$ echo $PATH
/opt/riscv/bin:/home/swetha/.local/bin:/home/swetha/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

jim-wilson, 2021-02-19:
I don't know anything about sel4.  If you are having a problem with that, you should probably ask the sel4 folks.

You didn't show the gcc command that failed.  There should be a /usr/include/features.h in the sysroot and the compiler should have found it.  Try debugging the build a bit.  However, I would expect riscv-pk to be built by an embedded elf compiler not a linux compiler.  I don't know if using a linux compiler to build it can work.