riscv-software-src/riscv-tools #337: Question regarding the quick start

huanpass, 2021-02-21:
Hi, 

The quick start says export RISCV=/path/to/install/riscv/toolchain.  I am a beginner to RISCV. Can i ask does the toolchain here means the riscv-gnu-toolchain namely the compiler?  I wonder how i can set it up. It will be best if example provided. 

Many thanks!

jim-wilson, 2021-02-22:
Yes, you need a riscv64-unknown-elf toolchain like riscv-gnu-toolchain, or an equivalent one.  You can try building riscv-gnu-toolchain.  Or you can try downloading one.  There are some toolchain binaries on the SiFive web site on the software page.  Debian has a toolchain that might work, but I think you would need a very recent version of debian to get access to it.

RaoShri, 2021-03-10:
Hi Huanpass,
You would have added PATH to riscv toolchain while installing riscv-toolcahin : https://github.com/riscv/riscv-gnu-toolchain
You would have added this in your ./bashrc file.
Ex : I had installed it in /opt/riscv/bin/ so my command would be:
export RISCV=/opt/riscv/bin

Hope this helps.