riscv-software-src/riscv-tools #338: Issue with buildsh : autoreconf

RaoShri, 2021-03-11:
Hi,
I am getting the following error while trying to build riscv-tools. 

>$ ./build.sh
Starting RISC-V Toolchain build process

libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:42: installing 'build-aux/ar-lib'
configure.ac:37: installing 'build-aux/compile'
configure.ac:28: installing 'build-aux/config.guess'
configure.ac:28: installing 'build-aux/config.sub'
configure.ac:30: installing 'build-aux/install-sh'
configure.ac:30: installing 'build-aux/missing'
Makefile.am: installing './INSTALL'
libjaylink/Makefile.am: installing 'build-aux/depcomp'
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: 'AC_PROG_RANLIB' is rendered obsolete by 'LT_INIT'
configure.ac:22: installing './compile'
configure.ac:35: installing './config.guess'
configure.ac:35: installing './config.sub'
configure.ac:17: installing './install-sh'
configure.ac:17: installing './missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
Makefile.am:22: installing './mdate-sh'
Makefile.am:22: installing './texinfo.tex'
src/gnulib/Makefile.am:85: error: GL_GENERATE_LIMITS_H does not appear in AM_CONDITIONAL
src/gnulib/Makefile.am:131: error: GL_GENERATE_STDBOOL_H does not appear in AM_CONDITIONAL
src/gnulib/Makefile.am:154: error: GL_GENERATE_STDINT_H does not appear in AM_CONDITIONAL
autoreconf: automake failed with exit status: 1

How do I fix this?

jim-wilson, 2021-03-11:
Perhaps you need an older verison of automake.  Or maybe riscv-openocd is broken.  You might try talking to them directly instead of reporting problems here.

RaoShri, 2021-03-12:
I am sorry, but talk to whom directly?

jim-wilson, 2021-03-12:
github.com/riscv/riscv-openocd