riscv-software-src/riscv-tools #341: Support for half float

amithmath, 2021-03-30:
Is there any support for half precision floating point/bfloat in RISC-V gcc/g++?

jim-wilson, 2021-03-30:
The zfh extension for IEEE binary16 support isn't frozen yet, so not in the FSF GCC tree yet.  But there is support for it on branches with zfh in the name.  There is also support for it on the rvv-intrinsic branch as the Vector extension is expected to be used in combination with the zfh extension.  Just add _zfh to the -march option, e.g. -march=rv64gc_zfh to enable it.  It is also supported in recent versions of the SiFive toolchain.  And there is support in LLVM too, but I don't know the details as I don't follow LLVM development.

I'm not aware of any bfloat support as yet.

amithmath, 2021-03-30:
I think SiFive toochain, are you point to: https://github.com/sifive/freedom-tools ? But I am unable to locate ./configure in order to enable -march=rv64gc_zfh

jim-wilson, 2021-03-30:
You build the toolchain normally (or download a binary from our web site).  And then when you run gcc you add the -march=rv64gc_zfh or similar option to get half float support.

amithmath, 2021-03-30:
OK, I think rv64 means risc-v 64? I am using RISC-V32, I think in this case when I run gcc -march=rv32gc_zfh?

jim-wilson, 2021-03-30:
Yes.  Add _zfh to whatever arch you are currently using.