riscv-software-src/riscv-tools #342: Problem getting started RISCV

NervenCid, 2021-04-12:
Hello to all there i'm installing the tools for RISCV:

https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html

I installed the Tool chain and qemu and configured the path in this way according the documentation:

https://github.com/riscv/riscv-gnu-toolchain

        ```
export RISCV=/opt/riscv
export PATH=$PATH:$RISCV/bin
```

And after:

and run these commands:

`cd linux git checkout v5.4.0 make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- defconfig
`
But i got these errors:

`*** Default configuration is based on 'defconfig' scripts/Kconfig.include:35: compiler 'riscv64-unknown-linux-gnu-gcc' not found make[1]: *** [scripts/kconfig/Makefile:83: defconfig] Error 1 make: *** [Makefile:567: defconfig] Error 2`

I think that i successfully installed qemu but i am having problems with the other two packages (Linux and busybox due a the error above).

Thank you

johnwinans, 2021-04-12:
You need to remove the space in 'riscv64-unknown-linux-gnu- defconfig'

jim-wilson, 2021-04-12:
Ignore the commend about removing the space.

Did you build a riscv64-unknown-linux-gnu-gcc cross compiler?  Is it on your path?  What does "which riscv64-unknown-linux-gnu-gcc" say?  If it is on your path then that should print the pathname.  Does it work well enough to build a hello world program?  If you can't compile a hello world program then you won't be able to compile the linux kernel.

The guide doesn't seem to say anything about building the cross compiler.  If on a recent distro, you can just apt/dnf/yum/whatever install one.  Otherwise, you can build a toolchain from github.com/riscv/riscv-gnu-toolchain.  When configuring, you will need a --prefix= option to specify the install dir, and add --enable-linux to get a linux toolchain, then run make to build and install it.  Once you have that, you should then be able to build linux.

johnwinans, 2021-04-12:
woops!  :-D

jim-wilson, 2021-04-12:
FYI This is the best source of info that I know of for building an image from scratch to run on qemu.
https://github.com/carlosedp/riscv-bringup/tree/master/Qemu
He also has pages for unleashed and unmatched.

If you just want an image to boot, then it is simpler to download one than it is to build one.  You can download a Fedora image for instance.
https://fedoraproject.org/wiki/Architectures/RISC-V/Installing

NervenCid, 2021-04-13:
Thank you for your answer and sorry for my english

By i understood can i build from any Linux? i'm using ubuntu on WSL

Because i think that i already installed Qemu but Linux and Busybox is missing

jim-wilson, 2021-04-13:
You can download a linux image if you want.  I gave a link to the Fedora image.  Or you can build a linux image.  I gave a link to the CarlosEDP page which has good info.