riscv-software-src/riscv-tools #343: Quickstart failed

2536786901, 2021-06-10:
Hello !
When I use the ./build.sh command to install riscv-tools, an error occurs and the config.log shows as follows:
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find crt0.o: No such file or directory
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss
collect2: error: ld returned 1 exit status
configure:2412: $? = 1
configure:2450: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "RISC-V Proxy Kernel"
| #define PACKAGE_TARNAME "riscv-pk"
| #define PACKAGE_VERSION "?"
| #define PACKAGE_STRING "RISC-V Proxy Kernel ?"
| #define PACKAGE_BUGREPORT "Andrew Waterman"
| #define PACKAGE_URL ""
| /* end confdefs.h. */
|
| int
| main ()
| {
|
|;
| return 0;
|}
configure:2455: error: in `/home/qiaoguanjie/riscv-tools/riscv-pk/build':
configure:2457: error: C compiler cannot create executables
what should I do?

jim-wilson, 2021-06-10:
Your riscv64-unknown-elf compiler is broken.  Try building a hello world program with it, you will likely get the same error for missing crt0.o.  Where did your riscv64 compiler come from?  Did you install an OS package, download a tar file, or build it yourself?  If you built it yourself, do you still have the build dir and build log?  If so, you should look at the build log for errors.  If you installed an OS package, what OS are you running and what package did you install?  If you downloaded a tar file, where did you download it from?

2536786901, 2021-06-10:
thank you for your reply
I use git clone --recursive https://github.com/riscv/riscv-gnu-toolchain to download on the ubuntu virtual machine. The test can use the riscv64-unknown-elf-gcc command to compile the hello.c file. What is the problem?

jim-wilson, 2021-06-10:
Something went wrong when building the compiler.  The toolchain does an initial compiler build without libraries, a C library build, and then a full compiler build with libraries.  It looks like your build failed during the C library build.  So you have a compiler, but no C library, and no compiler libraries like libgcc or libstdc++.  You need to look at your toolchain build log if you have it.  If you still have the build dir, you can just run make again and see what happens.  If the build succeeded make won't do anything.  if the build did not succeed, then make will try to build the stuff that failed the first time.  Pipe the make output to a file, and then look for an error in the make output.