riscv-software-src/riscv-tools #344: collect2.exe: error: ld returned 1 exit status

nicolast0604, 2021-07-09:
C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../i686-w64-mingw32/bin/ld.exe: ../gnulib/import/libgnu.a(getrandom.o): in function `getrandom':
C:/msys64/home/nicol/riscv-gnu-toolchain/riscv-gdb/gnulib/import/getrandom.c:129: undefined reference to `BCryptGenRandom@16'
collect2.exe: error: ld returned 1 exit status
make[3]: *** [Makefile:1866: gdb.exe] Error 1
make[3]: Leaving directory '/home/nicol/riscv-gnu-toolchain/build-gdb-newlib/gdb'
make[2]: *** [Makefile:10078: all-gdb] Error 2
make[2]: Leaving directory '/home/nicol/riscv-gnu-toolchain/build-gdb-newlib'
make[1]: *** [Makefile:866: all] Error 2
make[1]: Leaving directory '/home/nicol/riscv-gnu-toolchain/build-gdb-newlib'
make: *** [Makefile:499: stamps/build-gdb-newlib] Error 2

nicol@LAPTOP-M1F3S9SG MINGW32 ~/riscv-gnu-toolchain

I already added the following according to the previous issues discussion, but it still happened.
**readline/readline/terminal.c to add "#define NEED_EXTERN_PC"**

nicolast0604, 2021-07-09:
After make clean and re build, it works but 
there is no 
riscv32-unknown-elf-gdb.exe 

Please help or hint, thanks a lot.

jim-wilson, 2021-07-09:
There is no toolchain in riscv-tools.  I'm assuming that you are building riscv-gnu-toolchain instread.

If you don't have gdb at the end, then the build failed.  This may not be obvious if you are doing a parallel make build.  If you run make again, you will see it doing work if the build failed, as it will try to rebuild the part that failed.

We do our windows builds on Ubuntu using cross compilers.  Building on windows should work but is not well tested or supported.

This problem looks like
    https://sourceware.org/bugzilla/show_bug.cgi?id=27657
but the person looking at the problem wasn't able to reproduce the build failure so it doesn't have a solution yet.  Maybe check if you have bcrypt package installed, and install it if missing, or uninstall it if present, to see if that helps.

There is another discussion of the problem here
    https://sourceware.org/pipermail/gdb/2020-September/048927.html
This suggests a possible gdb Makefile bug.  Not immediately obvious if anything was done to fix that.

This is really a gdb problem not a riscv-gnu-toolchain problem.  You might want to discuss it upstream with the FSF GDB maintainers.  If there is a fix upstream, we can backport it to riscv-gnu-toolchain.

nicolast0604, 2021-07-10:
Yes, after run with $make not make -j4, there are errors related to gdb on my MSYS2.

C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../i686-w64-mingw32/bin/ld.exe: C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../lib/libncursesw.a(lib_termcap.o):(.bss+0x4): multiple definition of `UP'; ../readline/readline/libreadline.a(terminal.o):C:/msys64/home/nicol/opentitan/riscv-gnu-toolchain/riscv-gdb/readline/readline/terminal.c:779: first defined here
C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../i686-w64-mingw32/bin/ld.exe: C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../lib/libncursesw.a(lib_termcap.o):(.bss+0x0): multiple definition of `BC'; ../readline/readline/libreadline.a(terminal.o):terminal.c:(.bss+0x58): first defined here
C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../i686-w64-mingw32/bin/ld.exe: C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../lib/libncursesw.a(lib_tputs.o):(.bss+0x6): multiple definition of `PC'; ../readline/readline/libreadline.a(terminal.o):terminal.c:(.bss+0x5c): first defined here
C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.3.0/../../../../i686-w64-mingw32/bin/ld.exe: ../gnulib/import/libgnu.a(getrandom.o): in function `getrandom':
C:/msys64/home/nicol/opentitan/riscv-gnu-toolchain/riscv-gdb/gnulib/import/getrandom.c:129: undefined reference to `BCryptGenRandom@16'
collect2.exe: error: ld returned 1 exit status
make[3]: *** [Makefile:1866: gdb.exe] Error 1
make[3]: Leaving directory '/home/nicol/opentitan/riscv-gnu-toolchain/build-gdb-newlib/gdb'
make[2]: *** [Makefile:10078: all-gdb] Error 2
make[2]: Leaving directory '/home/nicol/opentitan/riscv-gnu-toolchain/build-gdb-newlib'
make[1]: *** [Makefile:866: all] Error 2
make[1]: Leaving directory '/home/nicol/opentitan/riscv-gnu-toolchain/build-gdb-newlib'
make: *** [Makefile:499: stamps/build-gdb-newlib] Error 2

I verified this with another run on ubuntu 18.04, the gdb was generated successfully so I guess it is specific in MSYS2 and maybe there are some workaround need to be took.

Could you please give me any hint? Thanks a lot.