riscv-software-src/riscv-tools #345: Build fails

Saqlinahamad, 2021-07-14:
echo timestamp > s-match
g++ -std=gnu++11  -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/. -I../.././riscv-gcc/gcc/../include -I../.././riscv-gcc/gcc/../libcpp/include -I../.././riscv-gcc/gcc/../libcody  -I../.././riscv-gcc/gcc/../libdecnumber -I../.././riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././riscv-gcc/gcc/../libbacktrace   -o gimple-match.o -MT gimple-match.o -MMD -MP -MF ./.deps/gimple-match.TPo gimple-match.c
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
Makefile:1142: recipe for target 'gimple-match.o' failed
make[3]: *** [gimple-match.o] Error 4
make[3]: Leaving directory '/home/akhil/Saqlinahamad/SweRV_EH2/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
Makefile:4426: recipe for target 'all-gcc' failed
make[2]: *** [all-gcc] Error 2
make[2]: Leaving directory '/home/akhil/Saqlinahamad/SweRV_EH2/riscv-gnu-toolchain/build-gcc-newlib-stage2'
Makefile:955: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/akhil/Saqlinahamad/SweRV_EH2/riscv-gnu-toolchain/build-gcc-newlib-stage2'
Makefile:615: recipe for target 'stamps/build-gcc-newlib-stage2' failed
make: *** [stamps/build-gcc-newlib-stage2] Error 2

Saqlinahamad, 2021-07-14:
I compiled it another time.
It worked

closed 2021-07-14 by Saqlinahamad

jim-wilson, 2021-07-19:
gimple-match.c is a very large generated C file.  You perhaps ran out of system memory while trying to compile the file.  It looks like the kernel OOM (out-of-memory) Killer killed a compiler process.