riscv-software-src/riscv-tools #346: riscv-pk submodule might need to update to keep compatible with gcc.

Mastuko, 2021-08-16:
I met with the ERROR: 
```
../pk/pk.c: In function 'rest_of_boot_loader':
../pk/pk.c:139:3: error: both arguments to '__builtin___clear_cache' must be pointers
  139 |   __clear_cache(0, 0);
      |   ^~~~~~~~~~~~~~~~~~~
make: *** [pk.o] Error 1
```
when compiling RISCV-tools.
Seems that this issue is the result of being incompatibe to the current gcc version. The same problem report could be found in :[https://gcc.gnu.org/bugzilla/show_bug.cgi?id=100316](url).
Further more, the author fixed the bug and merged into riscv-pk:master:
[https://github.com/riscv/riscv-pk/pull/240/commits/6c2909fd2ff9fd4925e0e9ce47a9a392aa190e4a](url)
I am wondering that if riscv-tools should switch the pk submodule to a compatible branch which could resolve this problem.

jim-wilson, 2021-08-16:
riscv-tools is no longer maintained and is mostly useless.  But yes, it you want to make it possibly useful, then you need to update all of the submodules that haven't been updated in 2 years or so.  And then test it to make sure it still works.

Mastuko, 2021-08-16:
Thanks for the reply, I think it's still a good repo for the beginers to get in touch with RISCV-Toolchain.

closed 2021-08-16 by Mastuko